arduino/Arduino #8093: Use Serial, SerialUSB and Serial1 same time with M0

warriorfenixSM, 2018-10-12:
Hi,

i write a last issue #8072 and @facchinm answer me that SerialUSB and Serial as the "same".
My question is if is possible use Serial to communicate with UART sensor and Serial1 to send the data and debug with SerialUSB. Can you help me?
Do you have any examples?
I try to use, but when i try the serial send throw SerialUSB but not sends nothing throw hardware pins.

Thanks for all

per1234, 2018-10-12:
This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules. Be sure to post your code, using code tags (`</>` button on the toolbar).

closed 2018-10-12 by per1234

PaulStoffregen, 2018-10-12:
When you post on the forum, please show the program you've written, even if it is simple.  Do not ask people on the forum to guess.  Actually show what you've done.  Remember, you're asking strangers on the internet for help.  A few extra minutes to compose a good message with complete info that actually shows the effort you have made gives a much better impression and greatly improves the changes someone will be *able* to help, and will *want* to invest their time to help you.

Do this on the forum, not github.