arduino/Arduino #8097: analogwriteresolution

nchlsgh1, 2018-10-14:
Intel Galileo Gen 2 supports 12 bit resolution.

I have tried calling analogWriteResolution(12); but it says that it is not in the scope.

I have further searched here https://emutex.com/educational/71-getting-started-with-intel-galileo-gen-2
But not able to locate this ArduinoIO.h as well.

This is on IDE 1.8.7
Thanks!

per1234, 2018-10-14:
The function definition is here:
https://github.com/01org/corelibs-galileo/blob/master/cores/arduino/wiring_analog.c#L87-L90
but there is no declaration, thus the error.

I'm not sure whether this was intentional or a bug. If a bug, I don't know if there is any chance of getting it fixed since Intel killed off the Galileo quite some time ago.

The workaround described at https://emutex.com/educational/71-getting-started-with-intel-galileo-gen-2 of adding the declaration to ArduinoIO.h should work fine. If you need help with doing that you should post on the [Arduino Forum](http://forum.arduino.cc/) and we'll help you out.

nchlsgh1, 2018-10-14:
Managed to finally locate the file.. For me AnalogIO.h is found here:
C:\Users\{User Name}\AppData\Local\Arduino15\packages\Intel\hardware\i586\1.6.7+1.0\cores\arduino

Just tested and 12bit PWM resolution works. Might be a bug and there is very little documentation for this for the Galileo Gen2 as well... Anyone I can approach for this?

Thanks for quick response too!

closed 2018-10-14 by nchlsgh1