arduino/Arduino #8100: Command Line Build Failure

weswitt, 2018-10-15:
When compiling sketches from the command line it seems that the include file behavior is different. The sketch below compiles in the IDE but fails from the command line with the error: "itest:2:18: error: test.h: No such file or directory". This is a simple sketch to demonstrate the problem. I have itest.ino and test.h. One note is that removing the ESP8266WiFi.h include fixes it. It seems that there is some funky include path stuff happening.

The sketch is as follows:

#include <ESP8266WiFi.h>
#include "test.h"

void setup()
{
}

void loop()
{
}

matthijskooijman, 2018-10-16:
What command are you using to compile this sketch exactly? Can you enable verbose compilation in the preferences and then show the verbose output of both the IDE build and the commandline build?

weswitt, 2018-10-16:
After further investigation I've found that the CLI build works the first time, but subsequent builds fail with the error below.

Command: /home/wesw/dev/arduino-1.8.6/arduino --pref build.path=/home/wesw/dev/src/homeauto/tempsensor/itest/bin --verify /home/wesw/dev/src/homeauto/tempsensor/itest/itest.ino

Output from CLI build:
Picked up JAVA_TOOL_OPTIONS:
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
Build options changed, rebuilding all
itest:2:18: error: test.h: No such file or directory
 #include "test.h"
                  ^
compilation terminated.
exit status 1

matthijskooijman, 2018-10-16:
Seems you're running into a problem that I recently also spotted (see https://github.com/arduino/arduino-builder/pull/299#discussion_r225479179 and https://github.com/arduino/arduino-builder/pull/299). The problem is triggered because your build directory is inside the sketch directory, which is incorrectly handled by the build environment. For now, my recommendation would be to use a build path outside of your sketch directory, and to watch that PR for further discussion and a real fix.

Note that there might be a second bug here: This problem also needs the "Build options changed, rebuilding all" to trigger. From what you're saying, it sounds like you're just running the exact same command twice, so the build options should not actually have been changed. I've seen this before on a Windows machine of a colleague, where we were also passing --pref to change the build path and/or sketchbook path and/or preferences.txt path IIRC, so there might be a problem there. Might also be related to having the build dir inside the sketch dir.

weswitt, 2018-10-16:
That fixed the issue for me, thanks.

closed 2018-10-16 by weswitt