arduino/Arduino #8110: Wrong Board Update Message

CombiesGit, 2018-10-18:
After IDE Start it shows the Message "Für manche Ihrer **Boards** ist eine Aktualisierung verfügbar."
On every start.
Ok...
Klick on **Boards** shows "Arduino AVR Boards .... Version 1.6.22 INSTALLED"
No UPDATE Button, no chance to install or change to 1.6.23

But:
If i compile an *.ino File, the Builder uses the 1.6.23 Version.

facchinm, 2018-10-18:
@CombiesGit which IDE version are you using?

CombiesGit, 2018-10-18:
Win10 AMD 64Bit
IDE 1.8.7

facchinm, 2018-10-18:
Ok, it must be a misalignment bug. When you update the IDE it should recognize that there's a new bundled AVR core and prompt to remove the old one. If the user does not confirm the operation, the old one must be used (not the new one).
@cmaglie I thinks we can replicate it "easily", or check if this will be solved by `arduino-cli` integration.

per1234, 2018-10-18:
Closing as a duplicate of the already fixed https://github.com/arduino/Arduino/issues/7973.

You're using an outdated version of the Arduino IDE. Please always test using the [latest hourly build of the Arduino IDE](https://www.arduino.cc/en/Main/Software#hourly) before submitting an issue report so that you can be sure that it wasn't already fixed.

If you find this issue still occurs with the hourly build then you can comment here and I'll reopen.

closed 2018-10-18 by per1234