arduino/Arduino #8111: Compiller error

ZASto, 2018-10-18:
As of the moment JRE was updated, the compiller spits out error:

`lto1: internal compiler error: in lto_output_varpool_node, at lto-cgraph.c:624
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
lto-wrapper: fatal error: /home/zasto/.arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino2/bin/avr-gcc returned 1 exit status
compilation terminated.
/home/zasto/.arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino2/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino Nano.`

facchinm, 2018-10-18:
Hi @ZASto ,
it's not about jre but about the new compiler. Closing as duplicate of https://github.com/arduino/Arduino/issues/7949 , let's continue debugging the issue there.

closed 2018-10-18 by facchinm

per1234, 2018-10-18:
I think this is actually a duplicate of https://github.com/arduino/ArduinoCore-avr/issues/39. Perhaps both issues are related.