arduino/Arduino #8116: core_cm0plus.h Not Found on Windows (but not macOS) Compiling Whizoo Controleo3 ReflowWizard Example

jdmcs, 2018-10-19:
I tried to update the firmware of my Whizoo Controleo3 following their firmware update procedures (http://www.whizoo.com/update). On two different computers -- one macOS 10.14 and one Windows 10, I have the following items installed:

Arduino 1.8.7
Arduino SAMD Boards (32-bit ARM Cortex-M0+) 1.6.19
Controleo3-1.4 Library (http://whizoo.com/f/Controleo3-1.4.zip)

Arduino was installed by the Windows Installer method on Windows 10. On both platforms, the BSP was installed using the Board Manager, and the Controleo3-1.4 Library was installed by the ZIP file method.

On both computers, I opened the Controleo3 v1.4 -> ReflowWizard Example and then set the board to "Arduino/Genuino Zero (Native USB Port)". And then on both I clicked the Verify button.

Under macOS, the example compiled successfully. However, under Windows 10, the compile fails with the following error, which looks like the compiler was unable to find the file core_cm0plus.h:

Arduino: 1.8.7 (Windows 10), Board: "Arduino/Genuino Zero (Native USB Port)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\jdmcs\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\jdmcs\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\jdmcs\Documents\Arduino\libraries -fqbn=arduino:samd:arduino_zero_native -ide-version=10807 -build-path C:\Users\jdmcs\AppData\Local\Temp\arduino_build_447030 -warnings=none -build-cache C:\Users\jdmcs\AppData\Local\Temp\arduino_cache_680525 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\jdmcs\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=C:\Users\jdmcs\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.bossac.path=C:\Users\jdmcs\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0 -prefs=runtime.tools.bossac-1.7.0.path=C:\Users\jdmcs\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0 -prefs=runtime.tools.openocd.path=C:\Users\jdmcs\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.9.0-arduino6-static -prefs=runtime.tools.openocd-0.9.0-arduino6-static.path=C:\Users\jdmcs\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.9.0-arduino6-static -prefs=runtime.tools.CMSIS.path=C:\Users\jdmcs\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=C:\Users\jdmcs\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.arduinoOTA.path=C:\Users\jdmcs\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.0 -prefs=runtime.tools.arduinoOTA-1.2.0.path=C:\Users\jdmcs\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.0 -prefs=runtime.tools.CMSIS-Atmel.path=C:\Users\jdmcs\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.1.0 -prefs=runtime.tools.CMSIS-Atmel-1.1.0.path=C:\Users\jdmcs\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.1.0 -verbose C:\Users\jdmcs\Documents\Arduino\libraries\Controleo3-1.4\examples\ReflowWizard\ReflowWizard.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\jdmcs\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\jdmcs\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\jdmcs\Documents\Arduino\libraries -fqbn=arduino:samd:arduino_zero_native -ide-version=10807 -build-path C:\Users\jdmcs\AppData\Local\Temp\arduino_build_447030 -warnings=none -build-cache C:\Users\jdmcs\AppData\Local\Temp\arduino_cache_680525 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\jdmcs\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=C:\Users\jdmcs\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.bossac.path=C:\Users\jdmcs\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0 -prefs=runtime.tools.bossac-1.7.0.path=C:\Users\jdmcs\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0 -prefs=runtime.tools.openocd.path=C:\Users\jdmcs\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.9.0-arduino6-static -prefs=runtime.tools.openocd-0.9.0-arduino6-static.path=C:\Users\jdmcs\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.9.0-arduino6-static -prefs=runtime.tools.CMSIS.path=C:\Users\jdmcs\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=C:\Users\jdmcs\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.arduinoOTA.path=C:\Users\jdmcs\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.0 -prefs=runtime.tools.arduinoOTA-1.2.0.path=C:\Users\jdmcs\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.0 -prefs=runtime.tools.CMSIS-Atmel.path=C:\Users\jdmcs\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.1.0 -prefs=runtime.tools.CMSIS-Atmel-1.1.0.path=C:\Users\jdmcs\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.1.0 -verbose C:\Users\jdmcs\Documents\Arduino\libraries\Controleo3-1.4\examples\ReflowWizard\ReflowWizard.ino
Using board 'arduino_zero_native' from platform in folder: C:\Users\jdmcs\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.19
Using core 'arduino' from platform in folder: C:\Users\jdmcs\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.19
Build options changed, rebuilding all
Detecting libraries used...
"C:\\Users\\jdmcs\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10807 -DARDUINO_SAMD_ZERO -DARDUINO_ARCH_SAMD -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804d -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Zero\"" "-IC:\\Users\\jdmcs\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-IC:\\Users\\jdmcs\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS-Atmel\\1.1.0/CMSIS/Device/ATMEL/" "-IC:\\Users\\jdmcs\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.6.19\\cores\\arduino" "-IC:\\Users\\jdmcs\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.6.19\\variants\\arduino_zero" "C:\\Users\\jdmcs\\AppData\\Local\\Temp\\arduino_build_447030\\sketch\\ReflowWizard.ino.cpp" -o nul
In file included from C:\Users\jdmcs\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.1.0/CMSIS/Device/ATMEL/samd21/include/samd21.h:69:0,

                 from C:\Users\jdmcs\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.1.0/CMSIS/Device/ATMEL/samd.h:105,

                 from C:\Users\jdmcs\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.1.0/CMSIS/Device/ATMEL/sam.h:470,

                 from C:\Users\jdmcs\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.19\cores\arduino/Arduino.h:48,

                 from C:\Users\jdmcs\Documents\Arduino\libraries\Controleo3-1.4\examples\ReflowWizard\ReflowWizard.ino:6:

C:\Users\jdmcs\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.1.0/CMSIS/Device/ATMEL/samd21/include/samd21g18a.h:237:26: fatal error: core_cm0plus.h: No such file or directory

 #include <core_cm0plus.h>

                          ^

compilation terminated.

exit status 1
Error compiling for board Arduino/Genuino Zero (Native USB Port).

facchinm, 2018-10-22:
Hi @jdmcs ,
the file that is not being found should be inside `C:\Users\jdmcs\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0/CMSIS/Include/`. Can you take a look in that folder? If it's not there there may have been a problem in the update procedure; I suggest you to reinstall the core and make sure the file is correctly unzipped in there.

madichodikov, 2019-07-12:
I had the same issue and I just cleared the App data folder then it ended up working. Hope that can help if this still continues to be an issue!

per1234, 2019-09-16:
@jdmcs were you able to resolve the problem? Please take a minute to provide an update so we know whether we can close this issue.

DraconicDragon, 2019-10-09:
Deleted the ...\Arduio15 file fixed it for me.

WiddeA, 2020-03-15:
On Windows 10 I had to start Adafruit IDE as an administrator for the Boards Manager to install correctly.

closed 2022-10-11 by per1234