arduino/Arduino #8117: Cloudchip IoT Platform library for ESP32 and ESP8266

NarsimuluCinasi, 2018-10-19:
library link: https://github.com/cloudchip-tech/ciot-iot-lib.git

Hi, Please add the cloudchip iot library to arduino library manager

per1234, 2018-10-19:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

NarsimuluCinasi, 2018-10-20:
Thanks for the supporting, We have tagged and released the library.

cmaglie, 2018-10-21:
Done!

NarsimuluCinasi, 2019-03-05:
Released v2.0.0 for Cloudchip IoT Platform, 

Release link - https://github.com/cloudchip-tech/cloudchip_iot_platform_lib/releases/tag/2.0.0

per1234, 2019-03-05:
@NarsimuluCinasi the Arduino Library Manager indexer job will automatically pick up new releases within a couple hours of you creating them. In fact, if you check Library Manager you'll see that your 2.0.0 release is already available. So it's not necessary to request an update for every release.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-publish-a-new-release-once-my-library-is-in-the-list

cc-io, 2020-02-11:
Hello Team, Recently we have changed our repo git url. we requested to update our git repo link to below.
https://github.com/cloudchip-io/cloudchip-iot-arduino-sdk

cmaglie, 2020-02-17:
Done.

closed 2020-02-17 by cmaglie

NarsimuluCinasi, 2020-05-15:
Hi, git link not yet updated it seems and not reflecting the latest releases in Arduino IDE, Requested to replace the old repo link https://github.com/cloudchip-tech/ciot-iot-lib with new link https://github.com/cloudchip-io/cloudchip-iot-arduino-sdk

per1234, 2020-05-15:
@NarsimuluCinasi the repository URL has been updated. You can verify this in http://downloads.arduino.cc/libraries/library_index.json

reason your 2.0.3 release was rejected by the Library Manager indexer is because you forgot to update the `version` value in library.properties before making the tag.

The reason the releases since 2.0.3 have been rejected is because you changed the `name` value in library.properties. The library is pinned to the `name` value at the time of the library submission and any tag with a different `name` value is rejected by the indexer. If you want to change the library name, you need to make a request here for it to be manually updated.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#why-arent-releases-of-my-library-being-picked-up-by-library-manager