arduino/Arduino #8119: IDE does not lunch Ubuntu 18.10

danieleperot, 2018-10-20:
I downloaded the new IDE 1.8.7 for Linux x64.
I am using the Oracle version of JDK and JRE.
While trying to lunch the IDE, i got the following error:

`Picked up JAVA_TOOL_OPTIONS: 
java.lang.Error: Cannot load com.sun.java.swing.plaf.gtk.GTKLookAndFeel
	at javax.swing.UIManager.initializeDefaultLAF(UIManager.java:1351)
	at javax.swing.UIManager.initialize(UIManager.java:1459)
	at javax.swing.UIManager.maybeInitialize(UIManager.java:1426)
	at javax.swing.UIManager.getLookAndFeel(UIManager.java:492)
	at processing.app.linux.GTKLookAndFeelFixer.installGtkPopupBugWorkaround(GTKLookAndFeelFixer.java:79)
	at processing.app.linux.Platform.setLookAndFeel(Platform.java:44)
	at processing.app.Base.<init>(Base.java:239)
	at processing.app.Base.main(Base.java:145)
`
I got it fixed by changing a line in the arduino script from:
`JAVA_OPTIONS=("-DAPP_DIR=$APPDIR" "-Dswing.defaultlaf=com.sun.java.swing.plaf.gtk.GTKLookAndFeel"`
to
`
JAVA_OPTIONS=("-DAPP_DIR=$APPDIR")
`
I hope this can help somebody with the same problem I had!

facchinm, 2018-10-22:
Hi @danieleperot ,
thanks for reporting! Are you using any particular flavour of Ubuntu or is it the plain image?

facchinm, 2018-10-22:
I've been able to reproduce it with a clean 18.10 installation. The problem is related with 18.10 shipping without `libgtk2.0` installed by default; I'm pushing a workaround soon.

danieleperot, 2018-10-23:
I can confirm that installing libgtk2.0 does fix the issue.
Thank you very much!

commit 04587709 (2018-11-05): Fix LookAndFeel loader

Fix #8119
Close #8122
closed 2018-11-05 by cmaglie