arduino/Arduino #8121: [Library Manager] Please add AnyRTTTL

end2endzone, 2018-10-21:
Please add the AnyRTTTL library to the Library Manager:

https://github.com/end2endzone/AnyRtttl

Thanks.

cmaglie, 2018-10-21:
Done!

closed 2018-10-21 by cmaglie

end2endzone, 2018-10-21:
that was fast! thank you!

per1234, 2018-10-27:
It works for me:
[image]
You can try deleting the local library index files to make sure the Arduino IDE downloads a fresh copy. I seem to remember some problems in the past IDE versions where a corrupted download could block subsequent attempts:

1. Click the link at the line following **File > Preferences > More preferences can be edited directly in the file**.
2. Delete all files that have names starting with library.index.
3. Check Library Manager again to see if AnyRtttl now appears.