arduino/Arduino #8125: [Library Manager] Please add AM2320_asukiaaa

asukiaaa, 2018-10-23:
I wrote library to control AM2330 temperature and humidity sensor.
Could you add this to library list?

https://github.com/asukiaaa/AM2320_asukiaaa

Thank you.

per1234, 2018-10-24:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

asukiaaa, 2018-10-24:
Thanks for the comment.
I released 1.0.0.
https://github.com/asukiaaa/AM2320_asukiaaa/releases/tag/1.0.0

cmaglie, 2018-10-30:
done!

closed 2018-10-30 by cmaglie

asukiaaa, 2018-10-30:
Thank you for merging.