arduino/Arduino #8129: Incomplete Fritzing file for MKR WAN 1300 which throws up errors

Gerriko, 2018-10-24:
I downloaded the Fritzing part for the MKR WAN 1300 but it throws up errors when I try and open it. When opened in the Parts editor there is also no PCB footprint provided and this too throws up errors. As such, the Fritzing part is incomplete and therefore should not be included under within the documentation section. Please, could someone up the part.

Oretoh, 2018-12-12:
@Gerriko Did you manage to find another fritzing file for mkr 1300?

vanepp, 2019-01-12:
If someone would provide a pointer to where the broken Fritzing file is I'll look at fixing it. Someone is asking about a variety of these boards (MKR GSM/Zero/Vidor)  on the fritzing forums (which is also the best place to get help on fritzing parts) which is how I got here, but I don't (in a quick look) see any fritizng parts. 

Peter

per1234, 2019-01-12:
@vanepp here it is:
https://content.arduino.cc/assets/MKRWAN1300Fritzing.zip

Thanks for looking into this!

Each of the MKR boards have a Fritzing file link from the "Documentation" tab of their product page. The product pages are linked from here:
https://www.arduino.cc/en/Main/Products

vanepp, 2019-01-13:
@per1234  well the cause is fairly easy, the pcb svg inside the fzpz is empty. Since I'm working on a part for the Arduino MKR Vidor it should be easy enough to generate a pcb svg that will do the job, since I think the pinouts and form factor are pretty much identical.  I'll post it here as I don't know where it really should go as a pull request because google can't find a repo that contains it. 

Peter

per1234, 2019-01-13:
Thanks! There is no repository for the product pages content so attaching your file here is the best way to proceed.

Since you're making a MKR Vidor part also, feel free to post the file here, or a link to wherever you're publishing it online. Since Arduino has put Fritzing parts on the product pages for the other MKR boards, it would be nice to add one for the MKR Vidor as well.

vanepp, 2019-01-13:
Finally done. There was more wrong in there than I first saw, so I corrected all of it. The corrected part is posted in the Fritzing Forums here:

http://forum.fritzing.org/t/arduino-mkr-wan-1300/6317

When I finish the Vidor (which is more complex yet, as it is poorly documented as far as I can find). I'll post it there too. As noted in the post on Fritzing, if the Arduino folks would like to replace the part on the store site with this one, please feel free to do so. 

Peter

vanepp, 2019-01-15:
I just finished and posted the Arduino MKR Vidor 4000 part. It can be obtained from 

http://forum.fritzing.org/t/part-request-arduino-mkr-boards/6304/8

as before, if the Arduino folks would like to post it on their product page feel free to do so. 

Peter