arduino/Arduino #8135: [Library Manager] Please add ACAN2517 and ACAN2517FD libraries

pierremolinaro, 2018-10-28:
Hello,

Please, can you include my ACAN2517 and ACAN2517FD libraries (CAN driver for MCP2517FD CAN Controller):
https://github.com/pierremolinaro/acan2517
https://github.com/pierremolinaro/acan2517FD

Best regards,

Pierre Molinaro

per1234, 2018-10-29:
Each library in the Library Manager index must have a unique `name` value in library.properties. Both of your libraries have the same `name` value: `ACAN2517`. Please change the `name` value in one of the library.properties files and then either create a new tag/release or redo the existing one.

Once you've done so, please comment here so we'll know we can proceed.

pierremolinaro, 2018-10-29:
Sorry for the mistake. I have changed name to ACAN2517FD in the library of the same name, and make a new release.

Thanks,

Pierre Molinaro

cmaglie, 2018-10-30:
done!

closed 2018-10-30 by cmaglie

pierremolinaro, 2018-10-30:
Thank you very much !