arduino/Arduino #8138: [BUG?] Export compiled binaries does not save sources if they are modified.

stef-ladefense, 2018-10-29:
Hello everyone,

I often need to generate binaries for remote updates.

Except if I make a modification in the source to adapt some things and that I generate a binary directly (Ctrl + Alt + S), this one does not contain the modification because the source is not saved before the compilation, although modified (sign § on the tab).

Currently I am forced to run a check (Ctrl + R) before generating a binary (Ctrl + Alt + S).

I think it's an oversight right?

Stéphane

facchinm, 2018-10-29:
Hi @stef-ladefense .
I just tested this with IDE 1.8.7 and it actually compiles the modified sketch (with or without "Save when verifying or uploading" flag). The invocation of arduino-builder in both "verify" and "export" is the same so I don't see how it could happen.
Would you mind sharing a little POC to demonstrate the issue? Thanks

stef-ladefense, 2018-10-29:
hi @facchinm 
here I am I do
load a source, make a change.
export binaries

in the project directory, the ino always has its original date which is different from the compilation date of the binaries (at the moment)

i work with 1.8.7 in Win10 64

[image]

have modified source, make binaire, time stamp no change in source and flag not too
[image]

how does the compiler generate the binary if the source is not backed up before?

facchinm, 2018-10-29:
Don't worry about the date; every time you compile a sketch with unsaved content, the IDE creates a copy of the modified sketch in a temporary directory, compiles it and, if requested, copies the binary back to the original sketch folder.
If you enable "Verbose compilation" you can see the temporary folder being used on the first invocation of `arduino-builder`.
Hope it helps :slightly_smiling_face:

stef-ladefense, 2018-10-29:
yes it helps and I saw.
just why does not it save the source automatically as the check does?

facchinm, 2018-10-29:
Ah, now I understand the problem! Let me try to code a fix :wink:

stef-ladefense, 2018-10-29:
very good ! thank

closed 2018-11-27 by cmaglie