arduino/Arduino #8146: board reset upon clicking upload

lpdunwell, 2018-10-31:
currently, when clicking 'upload' the connected target board gets put into reset, then compillation begins.
in the event of compilation errors, the board was reset, but no upload took place.

a better solution would be to wait with resetting the target until it's positive that compiling/linking succeeded and new data can be uploaded.

matthijskooijman, 2018-10-31:
I've noticed this issue as well, I think it has been present for quite some while (at least a year, probably more).

matthijskooijman, 2018-10-31:
W00ps

facchinm, 2018-10-31:
Mmmmh, I don't think it happens on all boards, since the uploader explicitly waits for the compilation to end successfully before kicking in. Could it be related to serial monitor opening/closing?

matthijskooijman, 2018-10-31:
I suspect that something indeed opens the serial port when the compilation starts (closing the serial port / monitor should not cause a reset AFAIK). I can't remember if this only happens when the serial monitor is opened, but I think it happens even without the serial monitor opened.

cmaglie, 2018-11-05:
Instead I've never noticed this, could you post your setup so we can try to reproduce? In particular:
- board used
- operating system
- IDE version
- is the serial monitor is open when clicking upload?
- a minimal sketch to see it happens...

dochawk64, 2019-02-28:
This was just referenced at 
http://forum.arduino.cc/index.php?topic=600452.0

It seems that, at least withy chinese nanos (both blue & black), resets occurs between compile and burn, after verification, and then again.

*BUT* only iff the serial monitor is open.

Also, both opening and closing the serial monitor trigger.