arduino/Arduino #8152: ESP8266:library AsyncPing

akaJes, 2018-10-31:
https://github.com/akaJes/AsyncPing

per1234, 2018-10-31:
From https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format:

>Library names must contain only basic letters (A-Z or a-z) and numbers (0-9), spaces (), underscores (_), dots (.) and dashes (-). It cannot start or end with a space, and also it cannot start with a number.

This is referring to the `name` field in library.properties. Your `name` value: `AsyncPing(esp8266)` contains characters not permitted for libraries in the Library Manager index: `(` and `)`.

Please change the name to a valid value and then either create a new release/tag or else redo the existing one.

Once that's done, please comment here so we'll know we can proceed.

per1234, 2019-02-12:
Since the necessary change was never made, I'm going to go ahead and close this issue. If you ever feel like switching to a compliant `name` value, feel free to comment here and I'll re-open.

closed 2019-02-12 by per1234