arduino/Arduino #8154: Compiler bug. If statement never executed. Adding filler function then executes it.

Samlof, 2018-10-31:
Removing line 107 in included file causes the surrounding if statement to never be executed. The related functions are ISR(TIMER1_COMPA_vect) for timer interrupt and loop().
[bug.ino.txt](https://github.com/arduino/Arduino/files/2535206/bug.ino.txt)

I have Arduino Ide 1.8.7, XDRduino Uno and using arduino ide for visual studio.

matthijskooijman, 2018-10-31:
I had a quick look at the code, didn't find anything obvious broken. Having said that, my experience is that this kind is more often a logic error than really a compiler bug. It seems you reduced the example a bit already, but can you make it smaller? When you say the if statement is "never executed", how do you come to that conclusion exactly?

PaulStoffregen, 2018-10-31:
The bug is with your code, not the compiler.  You need to use "volatile" for variables which are changed by interrupts.

Samlof, 2018-10-31:
[bug.ino.txt](https://github.com/arduino/Arduino/files/2536127/bug.ino.txt)

is an even shorter version. As it is state isn't changed into TORIGHT and only one led lights up. Adding PIND; into loop() causes it to be changed, curr_led starts increasing and all leds light up

Samlof, 2018-10-31:
I see Paul. Seems strange to me that adding in a line that reads an IO pin then causes that variable to turn into 'volatile'.

PaulStoffregen, 2018-10-31:
The reason it appear to be line 107 is the compiler "knows" variables in ordinary memory don't magically change.  It is allowed to make assumptions.  But the compiler refrains from such assumptions in certain cases, like when you call some other function that might have unknown side effects.  So adding line 107 is (probably) preventing the compiler from doing an optimization which "knows" curr_led is a variable in memory and no code could possibly have run which modifies it.

The compiler can't know when hardware interrupts occur.  So any variable modified by an interrupt *must* be declared with volatile, to tell the compiler that variable does magically change on its own sometimes.  Otherwise the compiler will (or might) apply very aggressive optimizations based on its knowledge that memory doesn't magically change itself and no other code could have run which would change it.

The compiler is correct.  Your code is not.

PaulStoffregen, 2018-10-31:
The factor disabling the compiler's optimizations not might be the fact it's a function call, but rather the special nature of the PIND register.  As I recall (it's been years since I've touched AVR assembly) those first several registers are access with special IO instructions, rather than being memory mapped.  That special instruction is probably a "memory barrier" in compiler lingo.

Samlof, 2018-10-31:
Yeah I found https://forum.arduino.cc/index.php?topic=418692.0 with 'volatile' as well. The compiler thinks that state isn't used anywhere in the code, so it removes the whole branch.

Samlof, 2018-10-31:
Thanks for the explanation, closing this now. Nothing you can really do about this problem in C.

closed 2018-10-31 by Samlof