arduino/Arduino #8157: Feature request :   Start user program before compiling to get some variables from outside IDE

belial11, 2018-11-01:
Add pls possibility to start user program before compiling like it done at most IDE.
Need this to automate the version control (for example to create header with current date/time by .vbs script.
Example to use :  versionfile-gen.vbs version.c :
<CODE>
'[Begin versionfile-gen.vbs]
On Error Resume Next
Dim Args
Set Args = WScript.Arguments

if (Args.Count < 1) then
  MsgBox "Error generate version file." + vbCrLf + "Usage:" + vbCrLf + "version-gen.vbs <filename>"
else
  'receive the output filename
  verfilename = Args(0)
  'receive date/time
  vYear  = CStr(Year  (Date))
  vMonth = CStr(Month (Date))
  if Len(vMonth) < 2 then
     vMonth = "0"+vMonth
  end if
  vDay    = CStr(Day    (Date))
  if Len(vDay) < 2 then
     vDay = "0"+vDay
  end if
  vHour  = CStr(Hour  (Time))
  if Len(vHour) < 2 then
     vHour = "0"+vHour
  end if
  vMinute = CStr(Minute (Time))
  if Len(vMinute) < 2 then
     vMinute = "0"+vMinute
  end if
  vSecond = CStr(Second (Time))
  if Len(vSecond) < 2 then
     vSecond = "0"+vSecond
  end if
  version_full = "const char version_full [] = ""GL"+vDay+"."+vMonth+"."+vYear+" "+vHour+":"+vMinute+":"+vSecond+""";"
  version_logo = "const char version_logo [] = ""SW: "+vDay+vMonth+Right(vYear, 2)+vHour+"."+vMinute+""";"
  
  Set FSO = CreateObject("Scripting.FileSystemObject")
  Set FileOutStream = FSO.OpenTextFile(verfilename, 2, true, 0)
  FileOutStream.Write version_full + vbCrLf
  FileOutStream.Write version_logo + vbCrLf
end if
'[End versionfile-gen.vbs]
</CODE>

per1234, 2018-11-01:
It sounds like maybe what you're asking for is already provided by platform.txt's `recipe.hooks.sketch.prebuild.NUMBER.pattern`:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification#pre-and-post-build-hooks-since-ide-165

Does that meet your requirements?

belial11, 2018-11-02:
@per1234 Thanks for push me in right direction, It works!!!

<code>
##------------------ My special wish----------------------------------------------------------------------------------------
##---------------------------------------------------------------------------------------------------------------------------
recipe.hooks.sketch.prebuild.NUMBER.pattern=wscript.exe versionfile-gen.vbs version.c
</code>

closed 2018-11-02 by belial11