arduino/Arduino #8162: [Library Manager] Add PIDArduino

DonnyCraft1, 2018-11-01:
[Here](https://github.com/DonnyCraft1/PIDArduino) is my GitHub repo.

Thanks!

per1234, 2018-11-01:
Your library.properties file is misspelled library.properies and is missing required fields. Please make your library.properties file compliant with the Arduino Library Specification and then either create a new tag/release or else redo the existing one.

After that's done, please comment here so we'll know we can proceed.

More information:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

DonnyCraft1, 2018-11-01:
@per1234 I renamed the file, added the required fields and released v1.0.1 with the updated code.

Thanks for the fast response!

DonnyCraft1, 2018-11-01:
I couldn't figure out what the `architectures` field meant, so I set it to all (*). Do you have any documents explaining this?

Thanks again!

per1234, 2018-11-01:
The `architectures` field is used for two things:
- Determine priority when multiple libraries match an `#include` directive: Explicit match > wildcard match > mismatch.
- Display a warning when the library is compiled for a board with an architecture that doesn't match the list.

I have added this information to the library specification.

Board architecture is determined by the architecture folder name of the hardware package:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification#hardware-folders-structure

DonnyCraft1, 2018-11-02:
Thank you very much! This was useful to know.

DonnyCraft1, 2018-11-05:
Let me know if there are any further issues with my library. I'll fix it.

cmaglie, 2018-11-07:
done!

closed 2018-11-07 by cmaglie

DonnyCraft1, 2018-11-07:
Thank you very much!