arduino/Arduino #8163: MacOS IDE "Window" menu?

kosso, 2018-11-02:
I can't find this in previous issues, but if possible, it would be hugely useful to include the (very standard) "Window" menu bar option on macOS. (No idea if this is also an issue on Windows or Linux).  

This menu usually includes a "Bring All To Front" option as well as the ability to see which sketch windows are open and makes it easy to select one to work on. 

The current situation without this menu option makes life (and coding) incredibly frustrating when you have a few sketches open, and a browsers open, etc.

DonnyCraft1, 2018-11-02:
I can confirm that there is no "Window" menu bar option on the IDE for windows.
My current IDE version is 1.8.5, and this is the options currently on the menu bar (On my IDE, Windows 10)

* File
* Edit
* Sketch
* Tools
* Help

per1234, 2018-11-02:
Closing as duplicate of https://github.com/arduino/Arduino/issues/234

We welcome bug reports and feature requests but in the future please take the time to search for existing open and closed issues and pull requests before submitting an issue:
https://github.com/arduino/Arduino/issues?&q=
If you do find the issue has already been reported or a fix has been submitted, you are welcome to give it a "thumbs up" by clicking on the smiley face icon at the top right to show your support. Please only comment on the issue if you have additional information to add.

closed 2018-11-02 by per1234

kosso, 2018-11-02:
@per1234  Thank you for pointing me to that issue. 

But it was opened in 2012!!  
And has had only one comment since 2016. 

With all due respect, that a bit crazy.

And also why I probably couldn't find it.