arduino/Arduino #8165: MKR WiFi 1010 Serial Monitor

victorbramhall, 2018-11-03:
The IDE disconnects the Serial Monitor after Compiling and doesn't know which port it had. This makes it very difficult to troubleshoot. The Serial Port has to be re-selected after each compile.

facchinm, 2018-11-05:
Hi @victorbramhall , this is totally unexpected. Which OS / IDE version are you using?

victorbramhall, 2018-11-05:
IDE version 1.8.7 hourly 10/24/2018
Windows 10 (latest update)
Arduino MKR WiFi 1010

The ide forgets which port that was used during compile, and the port has to be re-selected. It is intermittent but very seldom remembers. The ide reports COMM serial port 17 not available.  I have never had a come 17 serial port.

Victor Bramhall

harounhajem, 2018-12-11:
This is the same [bug i have ](https://github.com/arduino/Arduino/issues/8226). The bug began like this, it jumped to different COM-ports after upload. Today the device is unrecognizable by windows and not usable any more. In other words, it's bricked.