arduino/Arduino #8168: Library update deletes additional files in sketch folders

Keux486, 2018-11-03:
Hi, I´m a newbie in Arduino´s world and I don't know if is normal, but recently I've updated some Libraries in IDE and some "backup" files that I store in the same sketch folder has disappeared, they had the *.txt extension because I used to copy and paste my sketches in text files to avoid opening a lot of IDEs. My *.ino files remain untouched. Any help would be really appreciated because I've been working in a project and those files were my only backups

DonnyCraft1, 2018-11-03:
> I've updated some Libraries in IDE

Can you be more specific about how you updated the libraries?
Did you download the updated libraries manually or through the library manager?

*I can't help you with this issue since you probably know more than me already*

Keux486, 2018-11-03:
It was through the library manager, also maybe could be useful, I modified
preferences file, adding the line to change the *.hex file generated after
been compiled to another folder and be able to simulate Arduino Uno using
Proteus.

I just look into my Arduino projects folder and nothing is there :( I used
to have

.../Arduino/Test (sketches and simulations)
.../Arduino/Documentacion (Documents and datasheets)
.../Arduino/SIMON (Main project)

I used to make my drafts in Test folder, when something was OK, I save into
SIMON folder, but now nothing is there, just "core" "libraries" "preproc"
"sketch" and "Test" folders but inside Test is just the *.ino file

per1234, 2018-11-04:
>I modified preferences file, adding the line to change the *.hex file generated after been compiled to another folder and be able to simulate Arduino Uno using Proteus.

I suspect this is what caused your problem. Please explain exactly what you did.

matthijskooijman, 2018-11-05:
@facchinm, I considered that, but AFAIR, removing the build folder *only* happens when none is set in the preferences and a temporary location was generated. If a preferences-set build folder is wiped, that could considered a bug (though, thinking about it, the "build options changed, wiping build dir" does exactly this, of course... I wonder if we should implement that one differently (by setting a global "rebuild all" flag, rather than actually deleting files...)

Keux486, 2018-11-05:
I did this:

1. Modify preferences file to generate *.hex file in another path adding
"build.path:.../Arduino/Test" line to use it for drafts as I said in
previous mail (everything worked well)

2. Update a library (honestly I don't know which one) through library
manager

Thank you so much for your support, afortunately I've recovered some
important files using Recuva, but those files that can't be recovered
Recuva reported that were overwritten (I guess due update process)

Another thing to point at, is that in .../Arduino/Test folder, after
updating only *.ino file were remaining but everything else were deleted, I
don't know if was due that folder was selected in build.path line