arduino/Arduino #8172: "Please select a port before upload" breaks port-less devices

matthijskooijman, 2018-11-05:
It seems that the new "Please select a port before upload"-message that was added in #7958 breaks uploads to port-less devices, such as the digispark boards. These devices use a custom uploader tool that uses a custom USB protocol through libusb rather than a serial port.

On a new install, there would be no need to select a serial port with these boards. However, the newly added message is still shown, and uploading is prevented. On a system without serial ports, where a port has never been selected before, this completely prevents users from uploading their sketches.

Note that I haven't confirmed this problem myself, this report is based on a report from a user in #arduino a few days ago. However, from how I understand the code works, I expect this issue exists as described.

To fix this, I guess the IDE should become aware of whether a board needs a serial port to upload. One easy way would be to see if the upload recipe uses the serial port name (the [digispark core does not](https://github.com/digistump/DigistumpArduino/blob/master/digistump-avr/platform.txt#L79), though that feels a little bit fragile.

@PaulStoffregen

per1234, 2018-11-05:
Closing as duplicate of https://github.com/arduino/Arduino/issues/8052

closed 2018-11-05 by per1234

PaulStoffregen, 2018-11-05:
This is on my list to fix.  At the moment I'm crazy busy bringing up new hardware.  Can't work on this (or much of anything else) right now.

I see the "Start 1.8.8" commit just recently merged.  Please ping me if 1.8.8 is likely to release within a week or two and I'll try to put the hardware stuff aside for a while to get this fixed.  As long as 1.8.8 is many weeks or months away, I really need to prioritize on this hardware project.  It's what actually pays the bills and lets me contribute to Arduino and libraries.

per1234, 2018-11-06:
AFAIK, the "Start 1.8.8" commit has no significance. They usually do the version bump immediately after each release but sometimes that gets forgotten. That's a problem because then the hourly build displays a confusing version number in the title bar and **About** dialog (e.g. `Arduino IDE 1.8.7 Hourly Build 2018/10/24 04:33` instead of the expected `Arduino IDE 1.8.8 Hourly Build 2018/10/24 04:33`). I noticed the bump hadn't happened after the 1.8.7 release and so I submitted that PR. I have no insider knowledge about release schedules but I don't think that PR being merged is any indication since it's a simple, obvious change that there's no reason to wait on.

Thanks for all your contributions to the Arduino project PaulStoffregen. I think that your "Please select a port before upload" is a very valuable feature that will help a lot of beginners. I'm looking forward to a time when the posts about `avrdude: ser_open(): can't open device "\\.\COM1": The system cannot find the file specified.` no longer pop up on the forum several times a week.