arduino/Arduino #8174: Illegal reflective access by processing.app.linux.GTKLookAndFeelFixer

mMerlin, 2018-11-06:
Environment:
* 4.18.16-300.fc29.x86_64
* Arduino 1.8.5

This is a fresh load from scratch of Fedora 29 from Fedora-Workstation-Live-x86_64-29-1.2.iso.
```
sudo dnf upgrade
flatpak remote-add --if-not-exists flathub https://flathub.org/repo/flathub.flatpakrepo
flatpak install flathub cc.arduino.arduinoide
sudo usermod -a -G dialout «…»
```
reboot
```
$ flatpak run cc.arduino.arduinoide 
Picked up JAVA_TOOL_OPTIONS: 
Gtk-Message: Failed to load module "pk-gtk-module"
Gtk-Message: Failed to load module "canberra-gtk-module"
Gtk-Message: Failed to load module "pk-gtk-module"
Gtk-Message: Failed to load module "canberra-gtk-module"
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by processing.app.linux.GTKLookAndFeelFixer (file:/app/Arduino/lib/arduino-core.jar) to field com.sun.java.swing.plaf.gtk.GTKLookAndFeel.styleFactory
WARNING: Please consider reporting this to the maintainers of processing.app.linux.GTKLookAndFeelFixer
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
```

The app still runs correctly, as far as I can tell.  Currently only a warning, but the messages say it will be denied in a future release.  Of flatpak I assume.

mMerlin, 2018-11-06:
I missed that the flatpak was not part of the official software.  Repeating the issue information for flatpak maintainers https://github.com/flathub/cc.arduino.arduinoide/issues/5

A6GibKm, 2020-06-25:
I updated the flatpak app to use the latest version (1.8.13) and the issue persists, the only 'unknown' modification is that we are using java from opejdk11. It seems this is a java issue and it might be interesting to have it in mind.

jackaranda7691, 2020-12-25:
[
[image]
](url)

jackaranda7691, 2020-12-25:
I'm new to this and first time try at being an Arduino user. I don't understand much of the error messages but your thread seems relevant.  Help! ??

jackaranda7691, 2020-12-25:
processing.app.linux.GTKLookAndFeelFixer

I noted that permissions may be a problem.  However the permission Checker box never appeared and I don't know how to add myself to dialout, (which currently doesn't show me, just ---).