arduino/Arduino #8175: Online docs - tutorial/build-in examples - DigitalReadSerial -

tomaszc, 2018-11-06:
In online documentation page https://www.arduino.cc/en/Tutorial/DigitalReadSerial there are two errors:
1. circuit image shows not proper pin-2 connection to a breadboard. IMO it should be connected to the same leg of a pushbutton as a pull-down resistor to stabilize current when the pushbutton is not pressed.
2. In description there is a paragrap regarding LED blinking while on the circuit image and it's description there is nothing regarding LED

> If you disconnect the digital i/o pin from everything, the LED may blink erratically. This is because the input is "floating" - that is, it doesn't have a solid connection to voltage or ground, and it will randomly return either HIGH or LOW. That's why you need a pull-down resistor in the circuit.

per1234, 2020-05-23:
>circuit image shows not proper pin-2 connection to a breadboard. IMO it should be connected to the same leg of a pushbutton as a pull-down resistor to stabilize current when the pushbutton is not pressed.

The pins directly across from each other on a tactile switch are always connected, so there is nothing wrong with the diagram as far as the circuit goes. However, we might consider this an indication that the diagram would be easier to understand if it was changed to not rely on the internal connections of the tactile switch (which are not documented in the tutorial).

per1234, 2020-10-01:
>In description there is a paragrap regarding LED blinking while on the circuit image and it's description there is nothing regarding LED

This part of the report has now been resolved. Thanks so much for reporting it tomaszc!

The change to a more obvious breadboard configuration in the Fritzing diagram still needs to be done so I leave this issue open.