arduino/Arduino #8180: No driver on arduino nano

TheItalianGuy, 2018-11-07:
My arduino nano haven't any driver. I checked, it says "unknown device". How can I do?

matthijskooijman, 2018-11-07:
Is this an official Arduino Nano? Then the (FTDI) driver should be installed when installing the Arduino IDE. If it is a clone or counterfeit (for which this github does not provide support), it likely has a CH340G or similar USB chip, for which the driver must be found elsewhere.

matthijskooijman, 2018-11-07:
From your response, I take it you're not using an official Arduino board, so I'm closing this issue.
I can't tell you where to find the driver (I use Linux, which supports all these chips out of the box), but I suggest you Google for it, or look or ask on the Arduino forums. Or contact the supplier of your board, which should be your first point of contact for support requests.

closed 2018-11-07 by matthijskooijman