arduino/Arduino #8182: ardunio-builder crashes when compiling for "Optiboot on Mega 1280"

itopsuk, 2018-11-08:
Arduinio IDE 1.8.8, 1.8.9 & also 1.9 Beta showed me the same SIGSEV when trying to compile a simple sketch for "Optiboot on Mega 1280"

The current selected board needs the core 'arduino:arduino' that is not installed.
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x10 pc=0x1190c3c]

goroutine 1 [running]:
github.com/arduino/arduino-builder.(*SetupBuildProperties).Run(0x1860e50, 0xc420068400, 0x1860e50, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/setup_build_properties.go:98 +0x11bc
github.com/arduino/arduino-builder.(*ContainerSetupHardwareToolsLibsSketchAndProps).Run(0x1860e50, 0xc420068400, 0x1860e50, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/container_setup.go:63 +0x41d
github.com/arduino/arduino-builder.runCommands(0xc420068400, 0xc42004fda0, 0x3, 0x3, 0xc42004fd01, 0x10c0e95, 0xc420093790)
	/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/builder.go:207 +0xe5
github.com/arduino/arduino-builder.(*ParseHardwareAndDumpBuildProperties).Run(0xc42004fe00, 0xc420068400, 0x66, 0x7ffeefbffccb)
	/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/builder.go:195 +0xee
github.com/arduino/arduino-builder.RunParseHardwareAndDumpBuildProperties(0xc420068400, 0x66, 0x7ffeefbffccb)
	/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/builder.go:228 +0x35
main.main()
	/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/arduino-builder/main.go:396 +0x96a
/Applications/Arduino.app/Contents/Java/arduino-builder returned 2

Error compiling for board Optiboot on Mega1280.

facchinm, 2018-11-08:
Hi @itopsuk,
whould you mind sharing the exact sequence you are following to compile optiboot? Are you adding https://github.com/Optiboot/optiboot/releases/download/v8.0/package_optiboot_optiboot-additional_index.json to additional json index?

itopsuk, 2018-11-08:
Added the following as an additional JSON and the problem has been resolved

https://github.com/Optiboot/optiboot/releases/download/v8.0/package_optiboot_optiboot-additional_index.json

closed 2018-11-08 by itopsuk

facchinm, 2018-11-08:
Great, I'm a bit concerned anyway about the `nil pointer dereference` (it would be nice to reproduce it).
Would you mind writing down two lines about how you were getting the crash even if you solved it :smile:  ?

itopsuk, 2018-11-09:
Should be quite simple to reproduce.

Didn’t have the JSON in my preferences and then selected the board to be “Optiboot on Mega 1280”

Sketch-> Compile/Verify

Failed like this on a new or existing sketch.

Thanks for your help.

Regards Will.

facchinm, 2018-11-09:
Ok, but how did you get the “Optiboot on Mega 1280” entry? Did you add the files to `sketchbook/hardware` ?

itopsuk, 2018-11-09:
Thats a good question. I’ve been using Arduino on and off for sometime and it was just there
on the menu. Suspect its been there for sometime and I’ve only recently had to use it. Don’t recall removing anything
but I have been working on quite a big project and so anythings possible.

itopsuk, 2018-11-09:
Very weird, somebody has just emailed me a quick sketch to compile and the error has reappeared.

Looked in my preferences and the JSON had gone ! Don’t recall removing it.

Re-added it again and everything is now fine again. Thought it was only needed for finding the libraries in library manager?

The current selected board needs the core 'arduino:arduino' that is not installed.
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x10 pc=0x1190c3c]

goroutine 1 [running]:
github.com/arduino/arduino-builder.(*SetupBuildProperties).Run(0x1860e50, 0xc420068400, 0x1860e50, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/setup_build_properties.go:98 +0x11bc
github.com/arduino/arduino-builder.(*ContainerSetupHardwareToolsLibsSketchAndProps).Run(0x1860e50, 0xc420068400, 0x1860e50, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/container_setup.go:63 +0x41d
github.com/arduino/arduino-builder.runCommands(0xc420068400, 0xc420051da0, 0x3, 0x3, 0xc420051d01, 0x10c0e95, 0xc420095a00)
	/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/builder.go:207 +0xe5
github.com/arduino/arduino-builder.(*ParseHardwareAndDumpBuildProperties).Run(0xc420051e00, 0xc420068400, 0x9c, 0x7ffeefbffc68)
	/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/builder.go:195 +0xee
github.com/arduino/arduino-builder.RunParseHardwareAndDumpBuildProperties(0xc420068400, 0x9c, 0x7ffeefbffc68)
	/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/builder.go:228 +0x35
main.main()
	/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/arduino-builder/main.go:396 +0x96a
/private/var/folders/7m/4dqmyvgj34bd100ddmnjh1k8000101/T/AppTranslocation/A82637DE-3B68-459F-B51A-A2958DA3A7BC/d/Arduino.app/Contents/Java/arduino-builder returned 2

Error compiling for board Optiboot on Mega1280.