arduino/Arduino #8185: [Library Manager] Please add PVision IRCamera library

madhephaestus, 2018-11-09:
https://github.com/madhephaestus/PVision

This library communicates with the DF Robots IR camera and allows for user settable Wire busses, allowing for 2 or more IR Cameras at once.

per1234, 2018-11-09:
Leading zeros are not allowed in versions, as you currently have in your library.properties:
https://github.com/madhephaestus/PVision/blob/0.0.2/library.properties#L2
```
version=0.0.01
```

Please use a [semver compliant](https://semver.org/) version number in your library.properties, then either make a new release or else redo the existing release. You should also make sure that the version matches your release version (which it does not currently).

After that's done, please comment here so we'll know we can proceed.

madhephaestus, 2018-11-09:
I fixed this issue.

madhephaestus, 2018-11-12:
Pull requests merged

madhephaestus, 2018-11-14:
New version with corrections published.

cmaglie, 2018-11-14:
done!

closed 2018-11-14 by cmaglie

madhephaestus, 2018-11-14:
@cmaglie Thank you!