arduino/Arduino #8186: Export selected board port to arduino-builder as -prefs=~something.port=<selected_port>

garnould, 2018-11-09:
I am working on a sketch that as to be reconfigured manually when uploaded to one board (acting a a server) or another one (acting as a client).

I'd love to use preprocessor directives (#if) so that the sketch is automatically setup for the target, depending on the selected (board) port.

It looks that the would be straight forward to define the port in platform.txt (-D TARGET_PORT={~something.port}) if that information was made available.

Could IDE append this selected port info as a -prefs on cmdline while running arduino-builder ?

matthijskooijman, 2018-11-10:
A complication of doing this is that when changing the port, *all* code must be recompiled (since it can potentially change), which I think we normally want. I cannot see an obvious way to make this work for your sketch, without affecting the compilation performance of other sketches...

garnould, 2018-11-11:
You're right Matthijs, disabling compilation cache (when needed) affect compilation performance.

Let's say this feature would be for advanced users/dev (eg. working on modifying packages, my case) : this feature request would be totally transparent (as it only declares a new prefs that is used or not by developers) and permit to people like me to gain a lot of time when recompiling/uploading firmware to several different devices "at once".