arduino/Arduino #8187: Autoformat trashes fancy comments...

WestfW, 2018-11-10:
If I start with a nice (and relatively standard) command block like:

```
/****************************************************
 *  Test program for digitalWriteFast()             *
 *  November 2018, by Bill Westfield.               *
 *  This test sketch is in the Public Domain        *
 ****************************************************
 *                                                  *
 * This sketch isn't really meant to be run.        *
 *  Rather, you compile it and look at the code     *
 *  that has been created to confirm that it has    *
 *  the appropriate minimal instruction sequences.  *
 *                                                  *
 *  Test matrix:                                    *
 *                                                  *
 *     Arduino Core  Boards:                        *
 *                                                  *
 *      Uno (xx8)                                   *
 *      Micro (32u4)                                *
 *      Mega                                        *
 *                                                  *
```
And then use "autoforma", it deletes all of my carefully added "asterisk blocks" and leaves only:
```

/****************************************************
    Test program for digitalWriteFast()
    November 2018, by Bill Westfield.
    This test sketch is in the Public Domain
 ****************************************************
 *                                                  *
   This sketch isn't really meant to be run.
    Rather, you compile it and look at the code
    that has been created to confirm that it has
    the appropriate minimal instruction sequences.
 *                                                  *
    Test matrix:
 *                                                  *
       Arduino Core  Boards:
 *                                                  *
        Uno (xx8)
        Micro (32u4)
        Mega
 *                                                  */
```

per1234, 2018-11-10:
This is caused by the [`remove-comment-prefix` Artistic Style option](http://astyle.sourceforge.net/astyle.html#_remove-comment-prefix):
https://github.com/arduino/Arduino/blob/799879e8ae9f0cb76c7a76f4e945347b92d6352d/build/shared/lib/formatter.conf#L31

Related issue: https://github.com/arduino/Arduino/issues/3941

WestfW, 2018-11-11:
I would prefer that autoformat not touch comments, perhaps with the exception of indenting already indented comments to match the code indentation...

per1234, 2021-04-05:
Closing as wontfix because the current Artistic Style configuration is now thoroughly established and each user can configure it to their preference.

The new [ClangFormat](https://clang.llvm.org/docs/ClangFormat.html)-based auto formatter of Arduino IDE 2.x does not mess with the fanciness of block comments at all, so the future looks bright.

closed 2021-04-05 by per1234