arduino/Arduino #8191: Tabs (IDE1.8.5) + Dongsen Tech Pocket32

OlivierCROISET, 2018-11-11:
The tabs run correctly with ATMega328 or Arduino Uno.
When I compile the same program for a Dongsen Tech Pocket32 board, the parts in the others tabs cannot be compiled.  So all my program must be in 1 tab.

Have you some idea to solve this ?

facchinm, 2018-11-12:
@OlivierCROISET ,
can you paste a small POC that shows what is happening? The verbose compiler output should be enough.
The issue is probably about the preprocessor stopping at some point (probably due to Pocket32 core incompatibilities) and not declaring the right prototypes.

OlivierCROISET, 2018-11-15:
Hi, 
Sorry, the problem seems to be the ISD1700 Library. The error message is:

/*======= in the main tab of the Arduino IDE 1.8.5 ====== */
/==========================================================

#include "Ucglib.h"         // ILI9341 UCG library (TFT screen)
#include <SPI.h> 

//mapping of the ILI TFT screen      sclk  data   dc   cs  reset 
   Ucglib_ILI9341_18x240x320_HWSPI ucg(           22,   5,  21);

/*============= SETUP  ================== */

void setup()
{
   Serial.begin(115200);
// Start the TFT screen ILI9341  
   ucg.begin(UCG_FONT_MODE_SOLID);    
   ucg.clearScreen();
   ucg.setPrintPos( 20, 20);
   ucg.setColor(0, 0, 255);  // Blue
   ucg.print("Hello from the Setup in the main tab"); 
} 
                           
/*=========== MAIN LOOP ================== */
void loop()
{    
// display something usefull on the TFT screen 
   ucg.setPrintPos( 20, 40);
   ucg.setColor(0, 255, 0);      // Green
   ucg.print("Hello from the Loop in the main tab"); 

   SecondaryDisplay(); 
  }

/*======= in a SECONDARY tab of the Arduino IDE 1.8.5 ====== */
/===========================================================

void SecondaryDisplay()
{
// display something usefull on the TFT screen 
   ucg.setPrintPos( 20, 60);  
   ucg.setColor(255, 0, 0);       // Red
   ucg.print("Hello from the Loop in the secondary tab");  
}

/****************************************************************************/
The compiler error message is the Following ( I dont understand why the XPT2046 message in the 1st line remains here . . . XPT2046 was used in a previous sketch.)
The problem seems to be the ISD1700 registers declaration ? 
/****************************************************************************/
WARNING: Category '' in library XPT2046 is not valid. Setting to 'Uncategorized'      
C:\Users\papa\Documents\Arduino\libraries\ISD17xx-master\ISD1700.cpp: In constructor 'ISD1700::ISD1700(uint8_t)':

C:\Users\papa\Documents\Arduino\libraries\ISD17xx-master\ISD1700.cpp:67:3: error: 'SPCR' was not declared in this scope
   SPCR = (1<<CPHA) | (1<<CPOL) | (1<<DORD) | (1<<SPE) | (1<<MSTR);
   ^
C:\Users\papa\Documents\Arduino\libraries\ISD17xx-master\ISD1700.cpp:67:14: error: 'CPHA' was not declared in this scope
   SPCR = (1<<CPHA) | (1<<CPOL) | (1<<DORD) | (1<<SPE) | (1<<MSTR);
              ^
C:\Users\papa\Documents\Arduino\libraries\ISD17xx-master\ISD1700.cpp:67:26: error: 'CPOL' was not declared in this scope
   SPCR = (1<<CPHA) | (1<<CPOL) | (1<<DORD) | (1<<SPE) | (1<<MSTR);
                          ^
C:\Users\papa\Documents\Arduino\libraries\ISD17xx-master\ISD1700.cpp:67:38: error: 'DORD' was not declared in this scope
   SPCR = (1<<CPHA) | (1<<CPOL) | (1<<DORD) | (1<<SPE) | (1<<MSTR);
                                      ^
C:\Users\papa\Documents\Arduino\libraries\ISD17xx-master\ISD1700.cpp:67:50: error: 'SPE' was not declared in this scope
   SPCR = (1<<CPHA) | (1<<CPOL) | (1<<DORD) | (1<<SPE) | (1<<MSTR);
                                                  ^
C:\Users\papa\Documents\Arduino\libraries\ISD17xx-master\ISD1700.cpp:67:61: error: 'MSTR' was not declared in this scope
   SPCR = (1<<CPHA) | (1<<CPOL) | (1<<DORD) | (1<<SPE) | (1<<MSTR);
                                                             ^
C:\Users\papa\Documents\Arduino\libraries\ISD17xx-master\ISD1700.cpp:68:11: error: 'SPSR' was not declared in this scope
   dummy = SPSR;
           ^
C:\Users\papa\Documents\Arduino\libraries\ISD17xx-master\ISD1700.cpp:69:11: error: 'SPDR' was not declared in this scope
   dummy = SPDR;
           ^
C:\Users\papa\Documents\Arduino\libraries\ISD17xx-master\ISD1700.cpp: In member function 'uint8_t ISD1700::spi_transfer(uint8_t)':
C:\Users\papa\Documents\Arduino\libraries\ISD17xx-master\ISD1700.cpp:90:3: error: 'SPDR' was not declared in this scope
   SPDR = data;
   ^
C:\Users\papa\Documents\Arduino\libraries\ISD17xx-master\ISD1700.cpp:91:12: error: 'SPSR' was not declared in this scope
   while (!(SPSR & (1<<SPIF)));
            ^
C:\Users\papa\Documents\Arduino\libraries\ISD17xx-master\ISD1700.cpp:91:23: error: 'SPIF' was not declared in this scope
   while (!(SPSR & (1<<SPIF)));
                       ^
exit status 1
Error compiling for board Dongsen Tech Pocket 32.

facchinm, 2018-11-15:
`Dongsen Tech Pocket32 ` is ESP32 based, while the library you are trying to use is AVR-only (`SPDR` and friends registers). You need to find a library for ISD17xx compatible with ESP32.

closed 2018-11-15 by facchinm