arduino/Arduino #8192: LOGIN: CSRF: OOPS, SOMETHING WENT WRONG, PLEASE TRY AGAIN.

Flavio-da-Brescia, 2018-11-11:
What is this error?

I use chrome to login to arduino create, since forever, on windows 10, lenovo pc. Yesterday I started getting this error as i try to login to open my sketches. 
I am in seriuos trouble here. I cant move past the sign in screen

gvarisco, 2018-11-12:
Hey @Flavio-da-Brescia, thanks for contacting us. I wonder if any of your Chrome extensions is play a game here and troubling with the session cookie(s). Could you please remove your cookie set for create.arduino.cc and try again? You can also give it a try by logging-in using Incognito mode. Do let us know!

Flavio-da-Brescia, 2018-11-12:
Ciao Gianluca,

I tried incognito mode, and it works. I then tried removing all arduino
cookies, but it wasn't working. I tried removing all extensions and it now
works. Apparently one of them was my problem.

Thank you very much. Should I take any action regarding the issue I opnened
on github?

Have a nice day.

Flavio.

gvarisco, 2018-11-12:
Awesome, thanks for getting back to us so quickly. I'll go ahead and mark this as 'closed'. Feel free to re-open it if you encounter further issues.

Have a great day, Flavio!

closed 2018-11-12 by gvarisco

aster94, 2019-09-18:
Hello, same problem here, in my case the culprit extension was **Safe Torrent Scanner** from utorrent web which, i guess, a lot of people use this is the reason why I am writing this

molobox, 2019-11-14:
confirmed: Same problem **Registering** and **Signing** because of **Safe Torrent Scanner** installed by (I assume) an automatic update of the immediately uninstalled **uTorrent Web**!!

rkivanov, 2020-02-14:
Thank you aster94! I confirm that this works. It is enough to turn off Save Torrent Scanner form Google extensions (At the top right, click 3 vertical dots, then more tools, and then Extensions).