arduino/Arduino #8193: Java 11

salmanee, 2018-11-12:
Hey,

Did anyone manage to build and compile Arduino in JDK9+?

facchinm, 2018-11-12:
Hi @salmanee ,
the porting effort to Java9+ has not yet started, but any help in this would be very appreciated!

salmanee, 2018-11-12:
Thanks for your response @facchinm .. 
We are planning to do this as part of a course project. We managed to build Arduino but we couldn't run it (unable to locate the Java runtime environment). We suspect it has to do with setting up the JREs as those are not supported anymore in Java 11 and we need some help locating this in the build.xml file and replacing it properly. 
Any idea how this could be achieved?   

Thanks,
Sumaya

facchinm, 2018-11-13:
So, to start easily you can try building the "light bundle" package which doesn't contain the JRE (and falls back using the system one).
To achieve this, launch `ant build -Dlight_bundle=true` .
This also doesn't include the AVR core but it can be safely downloaded using Board Manager.
Then I'm sure there's a problem related with new Java versioning scheme (so a small version parser crashes when launching on Java9+) but most other stuff should be reasonably ok :wink: 
Let me know it it goes! It would be super cool to merge such a PR!

salmanee, 2018-11-13:
Thanks again for your response. I tried building Arduino using the "light bundle". Again the tool got **built** successfully but I still can't get it to **run** (It seems that its still looking for JREs as it states that it wasn't able to locate a Java Runtime Enviroment)

salmanee, 2018-11-14:
To further explain what we did:
- We installed Java11 and changed the $JAVA_HOME to point to the new jdk folder and managed to build Arduino successfully
- We then used: **jdeps** --list-deps app/**pde.jar** arduino-core/**arduino-core.jar** app/lib/***.jar** arduino-core/lib/***.jar** to list the jdk dependencies ==> it seems that JDK removed internal API/com.sun.image.codec.jpeg (we were not sure how to export or replace this dependency)  
- We used **jlinks** to produce a custom runtime for the application: jlink --no-header-files --no-man-pages --compress=2 --strip-debug --add-modules java.base,java.datatransfer,java.desktop,java.logging,java.naming,java.scripting,java.security.jgss,java.sql --output java-runtime
- Finally we ran: **java-runtime/bin/java** --module-path /Library/Java/JavaVirtualMachines/openjdk-11.0.1.jdk/Contents/Home/lib/ ==> we got the following error message
**java.lang.LayerInstantiationException: Package jdk.internal.jimage in both module java.base and module jrt.fs**

We are not sure how to proceed with this.. any help will be greatly appreciated!

facchinm, 2018-11-14:
Thanks for the explanation! Does it run without `jlink` phase, by just running the executable in `build/macosx/Arduino.app/...` ?
To bundle the full jvm you can build with
`ant build -Djava.net.preferIPv4Stack=true -Dplatform=macosx -DMACOSX_BUNDLED_JVM=$PATH/TO/TARGET/JVM`

salmanee, 2018-11-14:
It doesn't run with the executable. and bundling the full jam didn't work out either. 

It might have to do with the fact that one of the jar files contain an API the was removed in jdk9 (JDK removed internal API/com.sun.image.codec.jpeg)

salmanee, 2018-11-15:
I think I figured out the issue.. 

The reason why it's not able to locate a java runtime environment is because **com.apple.eawt** is removed from jdk9+ and should be replaced with **java.awt.desktop** ..

I can fix this issue by adding the following statement at compile time:
**-XaddExports:java.desktop/com.apple.eawt=ALL-UNNAMED** ==> which basically breaks the encapsulation of java.desktop module to access com.apple.eawt ..

my question is how can this statment be added at compile time for Arduino ?

facchinm, 2018-11-15:
`com.apple.eawt` is being used by `ThinkDifferent` class and here https://github.com/arduino/Arduino/blob/b7728beb52e9774c68e871495030a7de9e13d25c/app/src/processing/app/Base.java#L143 . The second snippet can be safely removed if you are not affected by the menubar bug, while the first file should be refactored (maybe with the help of Eclipse?)

tlk, 2020-02-03:
Here is an example from [Java 10: How to implement About, Preferences, and Quit menu items on MacOS](https://alvinalexander.com/source-code/java-macos-about-preferences-quit-handlers-desktop-awt) by Alvin Alexander: 

```java
import java.awt.*;
import javax.swing.*;

public class JavaAwtDesktop {

    public static void main(String[] args) {
        new JavaAwtDesktop();
    }

    public JavaAwtDesktop() {

        Desktop desktop = Desktop.getDesktop();

        desktop.setAboutHandler(e ->
            JOptionPane.showMessageDialog(null, "About dialog")
        );
        desktop.setPreferencesHandler(e ->
            JOptionPane.showMessageDialog(null, "Preferences dialog")
        );
        desktop.setQuitHandler((e,r) -> {
                JOptionPane.showMessageDialog(null, "Quit dialog");
                System.exit(0);
            }
        );

        SwingUtilities.invokeLater(() -> {
            JFrame frame = new JFrame("java.awt.Desktop");
            frame.setSize(new Dimension(600, 400));
            frame.setLocationRelativeTo(null);
            frame.setVisible(true);
        });
    }

}
```