arduino/Arduino #8196: OTA for devices outside of local network

Harvie, 2018-11-13:
I use arduino ide to program esp32 boards over OTA. But i can only do OTA for boards that are autodetected using mDNS. Can you please add possibility to manualy enter IP address of board, so i can use OTA for boards that are not in my local network (eg. in different subnet behind router)?

facchinm, 2018-11-15:
Hi @Harvie ,
implementing this would fundamentally modify the simplicity of "Ports" menu. My suggestion is to turn on "Verbose upload" from the preferences, copy the command that is being used to program the remote ESP32 (in a local network) and then replace the IP with the one you need.
Could it work for you?

Harvie, 2018-11-15:
> copy the command that is being used to program the remote ESP32 (in a local network) and then replace the IP with the one you need.

I've tried this, but it seems to me that OTA handler is not capable of comunicating over router. I've done tests and the routing of both TCP, UDP and ICMP traffic works fine, but still cannot get ArduinoOTA to work from different subnet across router. The ESP device responds to pings and HTTP requests, but OTA is not working.