arduino/Arduino #8199: is it possible to change the repository name which is already added Library Manager?

phpoc, 2018-11-15:
We have [the library](https://github.com/phpoc/arduino) which is already added to Library Manger.

When we requested to add our 1st library to Library Manager, we don't know we want to add another arduiono library.

But NOW we are going to add another arduino library to Library Manager.
Actually, we have plan to add 3rd, 4th, and so on.

Unfortunately, the repository name of our 1st library is arduino.
but the other libraries are also arduino library.
So the repository name of our 1st library may lead to confuse to people who use that library.

So we need to change this name, to distinguish our libraries clearly.

Are there any problem with people using our 1st arduino library via Arduino Library Manager
even if I change the repository name on GitHub
(of course this repository is our 1st library which is already added to Arduino Library Manager).

Thank you.

cmaglie, 2018-11-15:
If you rename the repository (but **not the library name** in library.properties) using the Github's Settings GUI there should be no "downtime" because github automatically redirects all request for the old repo to the new repo. Just tell us the new repo URL so we can change it in our indexer for completeness.

phpoc, 2018-11-16:
I am happy to hear your answer. Of course, the library name in library.properties will NOT be changed.
Anyway, I changed the repository name. Here is the new repo URL information.

(OLD) https://github.com/phpoc/arduino
(NEW) https://github.com/phpoc/arduino-Phpoc

I'd like you to apply this in your indexer.

Thank you for your support.

cmaglie, 2018-11-20:
done!

closed 2018-11-20 by cmaglie

phpoc, 2018-11-21:
I am really appreciate it.