arduino/Arduino #8200: [Library Manager] Please add MODI-library

hanj-kim, 2018-11-15:
https://github.com/LUXROBO/MODI-Arduino

thank you very much !

per1234, 2018-11-16:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

hanj-kim, 2018-11-16:
https://github.com/LUXROBO/MODI-Arduino/releases/tag/0.0.1

thanks for your feedback. I create a release 0.0.1 that matches the version value in my library.properties.file

cmaglie, 2018-11-20:
done!

closed 2018-11-20 by cmaglie