arduino/Arduino #8208: Switch maxed out on cases

Nightmystic1981, 2018-11-18:
I use Arduino IDE 1.8.7
If I use no more than 100-125 cases in a switch, no issues.
If I use more cases in a switch, lets say 150, I get error messages.
What is weird, is that the errors look different for Arduino and ESP8266.
To solve this, I make functions and place switches in the functions.
In the future I hope it will be possible to use as much cases I need without running into this weird issue.
The issue has nothing to do with memory shortage.

per1234, 2018-11-18:
>I get error messages

Please provide the error messages.

Nightmystic1981, 2018-11-19:
The code can be found here:
[https://gist.github.com/Nightmystic1981/633cebf570a81f71f75c66e5d62fd129](url)

Nightmystic1981, 2018-11-19:
This is the error I get (cut short, because of max. characters in a message):

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s: Assembler messages:

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:4796: Error: jump target out of range; no usable trampoline found

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:4796: Error: jump target out of range; no usable trampoline found

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:4796: Error: jump target out of range; no usable trampoline found

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:4796: Error: jump target out of range; no usable trampoline found

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:4796: Error: jump target out of range; no usable trampoline found

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:4796: Error: jump target out of range; no usable trampoline found

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:4796: Error: jump target out of range; no usable trampoline found

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:4796: Error: jump target out of range; no usable trampoline found

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:4796: Error: jump target out of range; no usable trampoline found

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:4796: Error: jump target out of range; no usable trampoline found

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:4796: Error: jump target out of range; no usable trampoline found

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:4796: Error: jump target out of range; no usable trampoline found

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:4796: Error: jump target out of range; no usable trampoline found

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:4796: Error: jump target out of range; no usable trampoline found

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:4796: Error: jump target out of range; no usable trampoline found

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:4796: Error: jump target out of range; no usable trampoline found

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:4796: Error: jump target out of range; no usable trampoline found

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:4796: Error: jump target out of range; no usable trampoline found

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:4796: Error: jump target out of range; no usable trampoline found

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:4796: Error: jump target out of range; no usable trampoline found

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:4796: Error: jump target out of range; no usable trampoline found

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:4796: Error: jump target out of range; no usable trampoline found

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:4796: Error: jump target out of range; no usable trampoline found

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:4796: Error: jump target out of range; no usable trampoline found

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:4796: Error: jump target out of range; no usable trampoline found

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:4796: Error: jump target out of range; no usable trampoline found

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:4796: Error: jump target out of range; no usable trampoline found

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:4796: Error: jump target out of range; no usable trampoline found

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:4796: Error: jump target out of range; no usable trampoline found

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:4796: Error: jump target out of range; no usable trampoline found

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:4796: Error: jump target out of range; no usable trampoline found

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:4796: Error: jump target out of range; no usable trampoline found

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:4796: Error: jump target out of range; no usable trampoline found

[... 42 lines cut ...]
C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:224223: Error: operand 2 of 'l32r' has out of range value '4294704892'

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:224257: Error: operand 2 of 'l32r' has out of range value '4294704872'

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:224295: Error: operand 2 of 'l32r' has out of range value '4294704844'

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:224326: Error: operand 2 of 'l32r' has out of range value '4294704824'

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:224349: Error: operand 2 of 'l32r' has out of range value '4294704808'

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:224383: Error: operand 2 of 'l32r' has out of range value '4294704788'

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:224417: Error: operand 2 of 'l32r' has out of range value '4294704768'

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:224455: Error: operand 2 of 'l32r' has out of range value '4294704740'

C:\Users\Portable\AppData\Local\Temp\ccO9AIA3.s:224482: Error: operand 2 of 'l32r' has out of range value '4294704720'

Continues with lots more or the above and then follows up with this:

Using library ESP8266WiFi at version 1.0 in folder: C:\Users\Portable\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi 
Using library Blynk at version 0.5.1 in folder: C:\Users\Portable\Documents\Arduino\libraries\Blynk 
Using library FastLED-3.2.0 at version 3.2.0 in folder: C:\Users\Portable\Documents\Arduino\libraries\FastLED-3.2.0 
exit status 1
Error compiling for board Generic ESP8266 Module.

Nightmystic1981, 2018-11-19:
On Arduino Uno:
SwitchCaseErrorTest2:240:6: error: total size of local objects too large

 void loop() 

      ^

lto-wrapper.exe: fatal error: C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc returned 1 exit status

compilation terminated.

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

Using library FastLED-3.2.0 at version 3.2.0 in folder: C:\Users\Portable\Documents\Arduino\libraries\FastLED-3.2.0 
exit status 1
total size of local objects too large

Nightmystic1981, 2018-11-25:
The same thing happens when I use IF statements.
The compiler seems to define the limit:
[https://stackoverflow.com/questions/20355948/maximum-number-of-cases-that-can-be-addressed-using-switch-statement](url)

llumero, 2020-10-11:
Hola.
Tengo un problema muy parecido al tuyo, por no decir que identico.
Queria saber si lo has podido solucionar y como.
Gracias

per1234, 2021-04-05:
Closing as off topic for this repository because this is a limit imposed by the compiler, which is not something Arduino has control over.

closed 2021-04-05 by per1234