arduino/Arduino #8210: AVR-Ada links broken

strangepete, 2018-11-18:
All of the links at https://playground.arduino.cc/Code/AVR-Ada are broken

The current links are based from:
http://sourceforge.net/apps/mediawiki/avr-ada/index.php?title=Main_Page

Where as the current working url appears to be:
https://sourceforge.net/p/avr-ada/wiki/Home/

per1234, 2018-11-18:
I have fixed all the broken links.

For future reference: The Arduino Playground is a publicly editable wiki. The Arduino community is responsible for the content of the Playground so it is not appropriate to create issue reports about it in this repository.  You are welcome to edit the Playground as you see fit. When you go to edit you will encounter a password page but that's just to prevent spam, the password is pre-filled. If you are unsure of how to fix an issue you find on the Playground or just want to discuss planned work with the community feel free to post on the [Arduino Forum.](https://forum.arduino.cc/index.php?board=21.0).

closed 2018-11-18 by per1234

strangepete, 2018-11-19:
Hi, thanks for the fix.

No where on that page is a button to edit, nor is it clear its a publicly editable wiki. When i clicked Contact Us at the bottom of the page, https://www.arduino.cc/en/Main/ContactUs it says errors in online documentation should be reported here.

per1234, 2018-11-19:
>No where on that page is a button to edit

I forgot to mention you need to be signed in to your arduino.cc account. After doing that, you'll see the edit button.

>nor is it clear its a publicly editable wiki

Well I think the text of the [Playground home page](https://playground.arduino.cc/) and the[ Participate page](https://playground.arduino.cc/Main/Participate) it links to makes it pretty clear. If you think the wording could be changed to make that more clear, you're welcome to make an edit.

>When i clicked Contact Us at the bottom of the page, https://www.arduino.cc/en/Main/ContactUs it says errors in online documentation should be reported here.

The layout of the Arduino website doesn't make an obvious visual distinction between the publicly editable Playground and the official documentation pages. So I can see how it would be easy to think that statement applied to the AVR-Ada Playground page.

But when you went to open an issue here, GitHub suggested that you read the Contributing Guidelines, which say:
https://github.com/arduino/Arduino/blob/master/CONTRIBUTING.md#1-before-using-the-issue-tracker
>Arduino Playground This is a publicly editable wiki. Please either make the edit yourself or create a post

strangepete, 2018-11-19:
I came to the AVR-Ada page out of the blue. I don't have an account, I didn't go to the home page, nor seen the participation page, nor did i see a message saying to read contribution guidelines (which i wouldn't have read anyway as I can barely see an issue report as 'contributing', which i would have assumed was for PRs)

I don't have time to chase broken links, so i figured i'd tell someone; if it were clear i needed to create an account, and read 4 separate pages across 2 sites, i wouldn't have bothered.

If you want people to let you know about simple issues, like broken links, you should provide a better, clearer interface to do so in the future. I edit Wikipedia all the time because its clear and easy; nothing on the playground looks or acts like any Wiki i've seen. Or just let people report issues here and just slap a label on it.

Thank you for taking the time to clarify all this; sorry to come off as irritated, although it is slightly irritating.

matthijskooijman, 2018-11-19:
Since we see issues about the playground regularly, I think it would be good to somehow more clearly mark it as a wiki. I was thinking about adding a note to the bottom on the sidebar, or perhaps change the "Participate" link into "Issues with playground pages? You can fix them." or something like that.

However, I couldn't actually edit this playground page, I couldn't get passed the anti-spam recpatcha and password. I sent a mail to the webmasters about this already.