arduino/Arduino #8212: Incorrect encoding of compiler log for Korean!

BrightLin, 2018-11-20:
Step:
- PC language:  Simplified Chinese.
- Installed Korean font support.
- Arduino IDE language: Korean.
- Verify!

Issue:
- unknown chars at compiler log window like "???? ???? ?? ?? 444 ???(1%)? ??. ?? 32256 ???."

Please see the following images. Thanks.

[image]
[image]
[image]

hanj-kim, 2018-11-20:
would you add '-Dfile.encoding=UTF8' in  arduino.l4j.ini or arduino_debug.i4j.ini file ?

[image]

BrightLin, 2018-11-27:
@hanj-kim Yeah~ It works. Thank you

feikname, 2018-11-27:
Is there any possible issue with having this option as default? If not, I'd like to suggest to add it to launch parameters of the next version.

PR #8289 (2018-12-11): Fixed incorrect encoding of compiler log
This commit fixes #8212 by passing '-Dfile.encoding=UTF8' to the arduino.l4j.ini file (windows launcher).

This property define the file encoding (UTF-8) that is required.
commit 0b688569 (2018-12-11): Fixed incorrect encoding of compiler log

This commit fixes #8212 by passing '-Dfile.encoding=UTF8' to the
arduino.l4j.ini file (windows launcher).

This property define the file encoding that is required.
closed 2018-12-11 by facchinm