arduino/Arduino #8226: MKR Wifi 1010 Upload

harounhajem, 2018-11-22:
Every time i upload to your board the upload fail and windows start to update the board drivers for some reason. The driver are working and added before and the port is selected before and doubble checked by me. 

Is the board broken or your IDE broken? I have a feeling if this keeps up the board will eventually be bricked. 

[image]

facchinm, 2018-11-23:
Hi @harounhajem ,
what is happening is very very strange; the uploader program (`bossac`) seems to execute correctly so the reported error could be about it crashing after upload.
The serial port reenumeration is strange too but it could be related with some component installed in the PC (antivirus?)
May I ask you to enable verbose upload from File/Preferences, copy here the commandline (which will start with something like `C:\Users\yourusername\Arduino15\.....\bossac.exe`) and your pc specifications (Widnows 7 edition and patch version, installed antivirus) ?
Thanks!

harounhajem, 2018-12-11:
I now did an successful upload of an example sketch and directly afterwards the device again got disconnected, changed COM port and  this time it never came back up as listed device. Now when i connect the device **_windows doesn't recognize_** it and says something like; _The usb-device is not recognized, recommendation: try to reconnect if this doesn't change your device is faulty._ 

I had a feeling it would end up like this since the first error turned up. I only bought this device as an assurance for not dealing with any low-level bugs so i could carry on with my development. 

Do you have any suggestions? Should i send it in? 

I'm running **Windows 10 Home** with **Arduino IDE 1.8.7** and no anti virus besides the windows builtin anti virus system. 

[image]

_it's in Swedish, but i says unknown usb-device(the request for device description failed)_

facchinm, 2018-12-14:
May I ask you to try using another USB cable (or plug into another USB port)? If windows complains about descriptors it means that something is going fundamentally going wrong at an hardware level. It may also be the board itself, maybe the USB connector is not perfectly soldered; in this case I'd ask you to contact the support for a replacement.

harounhajem, 2018-12-14:
I've spoken to Store support about replacement and they told me to reset the device, with double tap on the reset button so the orange LED starts fade. Now it works again. But what i can see in verbose upload is that the COM-port is changing.  It is found in the windows device manager on COM3
````
Sketch uses 10952 bytes (4%) of program storage space. Maximum is 262144 bytes.
Forcing reset using 1200bps open/close on port COM3
PORTS {COM3, } / {} => {}
PORTS {} / {COM5, } => {COM5, }
Found upload port: COM5
````

per1234, 2018-12-14:
I'm glad to hear you got it working again! The port change is indeed the expected behavior. After the reset, the bootloader runs and this is enumerated to a different port. After the upload, the board should again appear at COM3.

closed 2018-12-14 by per1234