arduino/Arduino #8227: [Compiling] Can't compile on MacOSX

rfcabal, 2018-11-23:
Hi Guys

I'm having troubles compiling a simple sketch on Mac, I'm using Mojave versión and I have Arduino IDE 1.8.7.

Once I click on check or upload It gets freezed on "Compiling Sketch". I have tried to ran like sudo and nothing happen. 

Here are my logs, I'm using Java 1.8 has default. Hope you can help me.

Thank you!

```
[JmDNS(fe80-0-0-0-aede-48ff-fe00-1122-en5.local.).State.Timer] WARN javax.jmdns.impl.tasks.state.DNSStateTask - Canceler(fe80-0-0-0-aede-48ff-fe00-1122-en5.local.).run() exception 
java.io.IOException: No route to host (sendto failed)
	at java.net.PlainDatagramSocketImpl.send(Native Method)
	at java.net.DatagramSocket.send(DatagramSocket.java:693)
	at javax.jmdns.impl.JmDNSImpl.send(JmDNSImpl.java:1618)
	at javax.jmdns.impl.tasks.state.DNSStateTask.run(DNSStateTask.java:131)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)
[fe80-0-0-0-1865-b6b4-2043-2516-en0.local..recover()] WARN javax.jmdns.impl.DNSStatefulObject$DefaultImplementation - Wait for canceled timed out: DNS: fe80-0-0-0-1865-b6b4-2043-2516-en0.local. [fe80:0:0:0:1865:b6b4:2043:2516%en0/fe80:0:0:0:1865:b6b4:2043:2516%en0] state: canceling 1 task: null
[fe80-0-0-0-1865-b6b4-2043-2516-en0.local..recover()] WARN javax.jmdns.impl.JmDNSImpl - fe80-0-0-0-1865-b6b4-2043-2516-en0.local..recover() Could not recover we are Down!
[fe80-0-0-0-aede-48ff-fe00-1122-en5.local..recover()] WARN javax.jmdns.impl.DNSStatefulObject$DefaultImplementation - Wait for canceled timed out: DNS: fe80-0-0-0-aede-48ff-fe00-1122-en5.local. [fe80:0:0:0:aede:48ff:fe00:1122%en5/fe80:0:0:0:aede:48ff:fe00:1122%en5] state: canceling 1 task: null
[fe80-0-0-0-aede-48ff-fe00-1122-en5.local..recover()] WARN javax.jmdns.impl.JmDNSImpl - fe80-0-0-0-aede-48ff-fe00-1122-en5.local..recover() Could not recover we are Down!

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/zreactor/Documents/Arduino/libraries -fqbn=arduino:avr:uno -vid-pid=0X2341_0X0043 -ide-version=10807 -build-path /var/folders/_b/8334gfj95qq4lk91rv1yl_3h0000gn/T/arduino_build_939925 -warnings=all -build-cache /var/folders/_b/8334gfj95qq4lk91rv1yl_3h0000gn/T/arduino_cache_484968 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -verbose /Users/zreactor/Projects/arduino/keypad_led/keypad_led.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/zreactor/Documents/Arduino/libraries -fqbn=arduino:avr:uno -vid-pid=0X2341_0X0043 -ide-version=10807 -build-path /var/folders/_b/8334gfj95qq4lk91rv1yl_3h0000gn/T/arduino_build_939925 -warnings=all -build-cache /var/folders/_b/8334gfj95qq4lk91rv1yl_3h0000gn/T/arduino_cache_484968 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -verbose /Users/zreactor/Projects/arduino/keypad_led/keypad_led.ino
Using board 'uno' from platform in folder: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr
Using core 'arduino' from platform in folder: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr
Detecting libraries used...
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10807 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard /var/folders/_b/8334gfj95qq4lk91rv1yl_3h0000gn/T/arduino_build_939925/sketch/keypad_led.ino.cpp -o /dev/null
```

facchinm, 2018-11-23:
Hi @rfcabal ,
the problem could be related with Gatekeeper/whatever filesystem protection is available in newer OSX.
A test could be trying to run 
```
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10807 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard /Users/zreactor/Projects/arduino/keypad_led/keypad_led.ino.cpp -o /dev/null
```
(after renaming `keypad_led.ino` to `keypad_led.ino.cpp`) from Terminal app, check if any warning is displayed and if the program ends or spins forever.

rfcabal, 2018-11-23:
> Hi @rfcabal ,
> the problem could be related with Gatekeeper/whatever filesystem protection is available in newer OSX.
> A test could be trying to run
>
> ``` [...]

Hi @facchinm 

I ran the command but nothing happen, not even a warning.

facchinm, 2018-11-23:
Ok, this is the normal (expected) behaviour. I wonder why it gets stuck if launched from inside the IDE...

rfcabal, 2018-11-23:
is the IDE using dependencies such as maven, ant, gradle?

facchinm, 2018-11-26:
Only at build time. When compiled and signed there's no external dependency (not even Java JRE since it's being shipped with the IDE itself).

rfcabal, 2018-11-30:
I tried to use https://create.arduino.cc/, brew in order to install arduino IDE and also this https://forum.arduino.cc/index.php?topic=570440.0. Nothing seems to work.

facchinm, 2018-12-03:
I have to say that there's something really weird going on in your mac... Create needs nothing more than a tiny agent installed and performs all the compilation remotely.
You are also the only user reporting these problems, which makes me think that they are related with your configuration (unless you are running some very particular Mojave version, like betas or similar).

rfcabal, 2018-12-03:
Here there is another guy who has the same problem, I installed Mojave when my system that could do it. I can't installed de plugin on my mac for create, It's eternally loading. [https://forum.arduino.cc/index.php?topic=575924.0](https://forum.arduino.cc/index.php?topic=575924.0)

rfcabal, 2018-12-08:
This got fixed with the IDE 1.8.8.

closed 2018-12-08 by rfcabal