arduino/Arduino #8229: java.lang.NullPointerException on 1.8.7

jaxxzer, 2018-11-24:
On archlinux:

```
$ pacman -S arduino
$ arduino
Picked up JAVA_TOOL_OPTIONS: 

(java:3021): Gtk-WARNING **: 16:47:47.119: Unable to locate theme engine in module_path: "adwaita",

(java:3021): Gtk-WARNING **: 16:47:47.122: Unable to locate theme engine in module_path: "adwaita",
java.lang.NullPointerException
	at processing.app.BaseNoGui.onBoardOrPortChange(BaseNoGui.java:679)
	at processing.app.Base.onBoardOrPortChange(Base.java:1313)
	at processing.app.Base.<init>(Base.java:276)
	at processing.app.Base.main(Base.java:145)
```

NicoHood, 2018-11-27:
I do not have this issue. Which java version did you install?
Try to update your system.

jaxxzer, 2018-11-27:
> May you test the Linux x86_64 build and check it it also exhibit the same problem?

What do you know, it works!

Re the pacman installation...

> Try to update your system. Edit: And reboot. Just in case.

Done so, no difference.

> Which java version did you install?

It was installed as a dependency of the arduino package.
```
$ java -version
openjdk version "1.8.0_192"
OpenJDK Runtime Environment (build 1.8.0_192-b26)
OpenJDK 64-Bit Server VM (build 25.192-b26, mixed mode)
```

PaulStoffregen, 2018-11-27:
I use 1.8.7 on Ubuntu 18.04 64 bit.  Works great.  Arduino.cc version has java bundled with it in the "java" folder.  Here's Java the version info:

    $ ./arduino-1.8.7/java/bin/java -version
    java version "1.8.0_171"
    Java(TM) SE Runtime Environment (build 1.8.0_171-b11)
    Java HotSpot(TM) 64-Bit Server VM (build 25.171-b11, mixed mode)

Alien1993, 2018-12-03:
I have the same issue, same version of Java, same version of the IDE. Installing the Linux x86_64 build solved the issue.

skuroedov, 2019-01-27:
Install [arduino-avr-core](https://www.archlinux.org/packages/community/any/arduino-avr-core/) package.

It should work.

NicoHood, 2019-01-29:
I think the problem is, that on a fresh install, with no board package installed, the IDE crashes. ArchLinux does not ship the default board package, because we package our own with upstream avr-gcc. But this package is not installed by default, as it is optional for those people who want to stick to the older arduino core. You can install it manually in the board manager.

The problem here is, that on the first startup none of both exists and the IDE crashes. There is also a bugreport with more details here:
https://bugs.archlinux.org/task/60378#comment175516

It seems that a special directory must be created first. This issue should be fixed in the IDE.

dianamazo, 2019-02-08:
please how Install arduino-avr-core package

skuroedov, 2019-02-13:
@dianamazo `# pacman -S arduino-avr-core`

NicoHood, 2019-02-25:
@facchinm People are contacting me over and over again that their IDE does not start. As explained earlier this is a bug in the IDE that should get fixed.

I do not want to make the avr core a normal dependency, as I had several reasons to make it optional and its no excuse to fix this issue. Its very likely that the fix is quite simple.

alfredopalhares, 2019-02-27:
I agree with @NicoHood the ide should not assume a default board is there.

facchinm, 2019-03-04:
@NicoHood there's a bit of logic that requires at least on platform to be installed. It could be fixed easily with a fake dependency that just injects a fake board; this would becomes a compulsory dependency, while marking `arduino-avr-core` incompatible with it (so pacman asks if you want to remove it to install the real core)

bluepoke, 2019-03-05:
I have this issue on Windows 10 with version 1.8.8 (installation package).
Reinstallation did not fix the problem.

dvukovic, 2019-03-06:
Win 7, Ver 1.8.8 same message.
What does this message even mean ?

What kind of problems are going to happen that I can not see ?

crenn, 2019-03-08:
Just experienced this on Win10 Pro x64, Version 1.8.8. Seemed to fix itself after I just went into the board manager once.

wallanpereira, 2019-05-05:
I solved the problem by installing arduino-avr-core

No Manjaro:

sudo pacman -S arduino-avr-core arduino

NicoHood, 2019-05-05:
@wallanpereira This has been mentioned already, but does not solve the problem. The Arduino team should fix this bug. All new users got this problem.

wallanpereira, 2019-05-05:
@NicoHood This is good, thanks for the info.

closed 2019-05-21 by facchinm

NicoHood, 2019-05-21:
Thanks for patching this long time bug! :-)

janbothatagger, 2020-08-30:
> I think the problem is, that on a fresh install, with no board package installed, the IDE crashes. ArchLinux does not ship the default board package, because we package our own with upstream [...]

Thanks, this fixed it for me.

ZoomRmc, 2020-12-02:
Still got the same (?) bug on fresh install, 1.8.13:
```
Picked up JAVA_TOOL_OPTIONS:
Set log4j store directory /home/xxx/.arduino15
java.lang.NullPointerException
	at processing.app.Base.rebuildProgrammerMenu(Base.java:1705)
	at processing.app.Base.<init>(Base.java:286)
	at processing.app.Base.main(Base.java:150)
```

matthijskooijman, 2020-12-03:
@ZoomRmc, you're seeing a different problem. The symptom (NullPointerException) is similar, but the location where it happened is totally different.

I'm pretty sure that you're seeing the problem that is fixed by #10922. Maybe you could try the version from that PR (see the comment by ArduinoBot) to see if that fixes your problem and report your findings in that PR? Thanks!

PR #8857 (2019-05-21): Fix NPEs if no hardware core is installed
Fixes #8229

@NicoHood can you check if this solves the (non-hard) dependency on `arduinocore-avr` with your build scripts?
commit d518d768 (2019-05-21): Fix NPEs if no hardware core is installed

Fixes #8229