arduino/Arduino #8240: [Library Manager] add WebUSB library

monteslu, 2018-11-27:
This library is the Arduino implementation of the WebUSB spec:  https://wicg.github.io/webusb/

The library can be found here:  https://github.com/webusb/arduino/tree/gh-pages/library/WebUSB

per1234, 2018-11-27:
Your repository does not comply with the requirements for inclusion in the Library Manager index:
- Library must be in the root of the repository, not in the `library/WebUSB` subfolder. I see that was already proposed in https://github.com/webusb/arduino/pull/15 but I think it also might well be a good idea to have a dedicated repository for the library only, then another one for the [webusb:avr hardware package](https://github.com/webusb/arduino/tree/gh-pages/hardware/webusb/avr).
- Repository must have a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

Once you have made these changes, please comment here so we'll know we can proceed.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

monteslu, 2018-11-27:
fair enough, we'll get that fixed up.  thanks @per1234

per1234, 2019-02-12:
Since the issues with the repository have still not been resolved after some time, I'm going to close this. Please comment here once you're ready to go and I'll re-open.

closed 2019-02-12 by per1234