arduino/Arduino #8253: [Library Manager] Please add Regexp library

coddingtonbear, 2018-11-30:
The commonly-used Regexp library was recently updated to be compatible with the Arduino Library Manager; would you mind adding it to the manager?

* https://github.com/nickgammon/Regexp

Cheers!

per1234, 2018-11-30:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

coddingtonbear, 2018-12-02:
Sorry about that, @per1234!  That tag is added now.

cmaglie, 2018-12-04:
Done!

closed 2018-12-04 by cmaglie