arduino/Arduino #8255: Fail to build IDE 1.8.8

JeanLucCoulon, 2018-11-30:
Hi,
I fail to build the IDE 1.8.8.
I run Debian sid / GNU Linux.
I get the following:

> 
> /usr/local/src/arduino/arduino-1.8.8/build/ ant dist
> Buildfile: /usr/local/src/arduino/arduino-1.8.8/build/build.xml
> 
> revision-check:
>      [echo] AVR Arduino core version in build.xml is: 1.6.23
>      [echo]     latest in package_index_bundled.json is: 1.6.23
>      [echo] Latest revision detected in shared/revision.txt is: 1.8.7
>      [echo]                      Revision in BaseNoGui.java is: 1.8.8
> 
> dist:
>     [input] Enter version number: [1.8.7]
> 1.8.8
> 
> build:
> 
> revision-check:
>      [echo] AVR Arduino core version in build.xml is: 1.6.23
>      [echo]     latest in package_index_bundled.json is: 1.6.23
>      [echo] Latest revision detected in shared/revision.txt is: 1.8.7
>      [echo]                      Revision in BaseNoGui.java is: 1.8.8
> 
> linux-checkos:
> 
> subprojects-build:
> 
> compile:
>      [copy] Copying 66 files to /usr/local/src/arduino/arduino-1.8.8/arduino-core/bin

**[...]

Then :** 

>      [copy] Copying /usr/local/src/arduino/arduino-1.8.8/arduino-core/src/processing/app/i18n/Resources_zh_TW.properties to /usr/local/src/arduino/arduino-1.8.8/arduino-core/bin/processing/app/i18n/Resources_zh_TW.properties
> 
> build:
>       [jar] Building jar: /usr/local/src/arduino/arduino-1.8.8/arduino-core/arduino-core.jar
> 
> compile:
>     [javac] Compiling 89 source files to /usr/local/src/arduino/arduino-1.8.8/app/bin
>     [javac] warning: [options] bootstrap class path not set in conjunction with -source 8
>     [javac] /usr/local/src/arduino/arduino-1.8.8/app/src/processing/app/EditorToolbar.java:158: error: cannot access Window
>     [javac]             touchBar.show(editor); 
>     [javac]                     ^
>     [javac]   class file for javafx.stage.Window not found
>     [javac] Note: Some input files use or override a deprecated API.
>     [javac] Note: Recompile with -Xlint:deprecation for details.
>     [javac] Note: Some input files use unchecked or unsafe operations.
>     [javac] Note: Recompile with -Xlint:unchecked for details.
>     [javac] 1 error
>     [javac] 1 warning
> 
> BUILD FAILED
> /usr/local/src/arduino/arduino-1.8.8/build/build.xml:136: The following error occurred while executing this line:
> /usr/local/src/arduino/arduino-1.8.8/build/build.xml:113: The following error occurred while executing this line:
> /usr/local/src/arduino/arduino-1.8.8/build/build.xml:161: The following error occurred while executing this line:
> /usr/local/src/arduino/arduino-1.8.8/app/build.xml:82: Compile failed; see the compiler error output for details.
> 

Regards

Jean-Luc

sandeepmistry, 2018-12-03:
> @sandeepmistry are you aware of any other library for touch bar that doesn't require javafx?

@facchinm unfortunately JTouchBar was the only Java TouchBar library I could find.

agdl, 2018-12-04:
```sudo apt-get install openjfx``` fixed it for me!

JeanLucCoulon, 2018-12-04:
I've openjfx, openjfx-source, liboenjfx-java and libopenjfx-ini installed.
Version on debian sid is 11+26-5

And I've the error reported here.

Jean-Luc

JeanLucCoulon, 2018-12-04:
BTW, I've miss-clicked the "close & comment" button and I dont know how to reopen

NicoHood, 2018-12-11:
Is it a build or runtime dependency?

JeanLucCoulon, 2018-12-12:
I failed to build it... So I've no idea about runtime.

matthijskooijman, 2019-01-13:
I was running into the same error, installing the `openjfx` package fixed the build for me.

Debian stable
javac 1.8.0_151
openjdk-8-jdk version  8u151-b12-1~deb9u1.

The OP is using Debian stable, maybe something changed there? Or maybe he's using a different (custom) version of java that is not finding the javafx packages installed through apt?

JeanLucCoulon, 2019-01-14:
I use Debian/sid, so the version of openjdk is 10/11/12... by choice.

- I've installed and activated 8, this doesnt allow to compile arduino IDE.
- Then I've installed openjfx and libopenjfx-java 8u11-b14 instead of the sid version (11.0.1).

This ALLOWS to build arduino IDE but the system is not consistent as libopenjfx-jini depends on libavcodec57 / libavformat57 but the version I've is 58 instead of 57 and there are MANY packages which depends on libavcodec58.

Anyway. The only way I have had to build arduino IDE was to use :  openjdk version 8 and libopenjfx-java 8u11
java 11/12 doesnt work, neither openjfx 11

Regards

Jean-Luc

ttveldhuis, 2019-01-21:
> > @sandeepmistry are you aware of any other library for touch bar that doesn't require javafx?
> 
> @facchinm unfortunately JTouchBar was the only Java TouchBar library I could find.

I hope to have removed any external dependencies in the next release. Will try to work on getting that out this week.

JeanLucCoulon, 2019-01-22:
There is a new version of openjfx on debian/sid (11.0.1+1-1 -> 11.0.2+1-1), with this version AND javac 8, it works (it compiles at least). 
But I've still the build error with javac 12

J-L

facchinm, 2019-01-22:
https://github.com/arduino/Arduino/pull/8417 targets the build issue with jfx (by removing the unneeded dependency). Test it if you can in a clean system and report the results there.
About Java12, any patch required to make the IDE build (and run) with java8+ is well accepted :slightly_smiling_face:

gvarisco, 2019-01-24:
I was able to compile the linux version on a clean Fedora system (javac 1.8.0_191) thanks to #8429.

giansalvo, 2019-02-17:
Hi,
I'm trying to compile Arduino 1.8.8 on macOS in order to add it to MacPorts. I'm using openjdk but there isn't openjfx package in MacPorts. Compile fails with the same error reported at the beginning of the thread. Is there a workaround to fix this issue? Thanks
```
build:
      [jar] Building jar: /Users/giansalvo/Downloads/Arduino-1.8.8/arduino-core/arduino-core.jar

compile:
    [mkdir] Created dir: /Users/giansalvo/Downloads/Arduino-1.8.8/app/bin
    [javac] Compiling 99 source files to /Users/giansalvo/Downloads/Arduino-1.8.8/app/bin
    [javac] /Users/giansalvo/Downloads/Arduino-1.8.8/app/src/processing/app/EditorToolbar.java:158: error: cannot access Window
    [javac]             touchBar.show(editor); 
    [javac]                     ^
    [javac]   class file for javafx.stage.Window not found
    [javac] Note: /Users/giansalvo/Downloads/Arduino-1.8.8/app/src/processing/app/Base.java uses or overrides a deprecated API.
    [javac] Note: Recompile with -Xlint:deprecation for details.
    [javac] Note: Some input files use unchecked or unsafe operations.
    [javac] Note: Recompile with -Xlint:unchecked for details.
    [javac] 1 error
```

gvarisco, 2019-02-18:
@giansalvo hey! We removed the openjfx dependency after releasing 1.8.8. While we work on releasing 1.8.9, could you build it against our `master` branch?

giansalvo, 2019-02-18:
@gvarisco I'll try and let you know.

giansalvo, 2019-02-18:
@gvarisco Here is what happen when I build against ```master``` branch: the build process is successfully completed but when I launch Arduino.app in ```build/macosx/work``` I get "Unable to load Java Runtime Environment" error.
These are the command used to build Arduino using openjdk8:
```
export JAVA_HOME=/Library/Java/JavaVirtualMachines/openjdk8/Contents/Home/
ant clean dist -Dplatform=macosx
```

On the other hand, if I use the official JDK from Oracle/Apple(?) to build the app, then Arduino.app run correctly. 
```
export JAVA_HOME=/Library/Java/JavaVirtualMachines/jdk1.8.0_131.jdk/Contents/Home/
ant clean dist -Dplatform=macosx
```

I believe there is a missmatch somewhre about the JRE version. Where can I check wich JRE is used at run time by the Arduino.app?

giansalvo, 2019-02-21:
The problem in MacPorts was fixed by using ```/usr/libexec/java_home``` to ```$JAVA_HOME```. Arduino will be available through MacPorts in the next update with ```sudo port install Arduino```.