arduino/Arduino #8257: Arduino-generated temp code throws errors: exit status 1 variable or field 'xxx' declared void

dsyleixa, 2018-12-02:
Arduino-generated temp code throws errors: 
exit status 1 variable or field 'xxx' declared void 

```
#include <foo.h>
#include <bas.h>
#include <other.h>
#include <Adisplay.h>
Adisplay display;

typedef struct  {
   int foo;
   char bas;
} list;

list mList00;

//  display Funktion
**void  displaylist(list mList) {   <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<**
    Serial.println(mList.foo);
    Serial.println(mList.bas);
}
```
// 
```
void setup() {
   mList00.foo=100;
   mList00.bas=3;

  displaylist(mList00);   
   //
}

```

Only if I put the struct definietion to the most upper top, then it compiles.
Which a bit weird, tbh.

dsyleixa, 2018-12-03:
ok, ftm I can't install a 1.8.7 version, it takes too much time and too many efforts, so for now this is a cutdown test code.
Is anyone has a new 1.8.7 he may try, otherways I'll try to test a 1.8.7 in a couple of days by my own:

/*
 *      using ESP_SSD1306 Adafruit library  
 * for a ssd1306 OLED Display  with the ESP8266 board
 *
 *********** This code is for the ESP8266 *********** 
 * 
*/

// I2C
#include <Wire.h>         // Incl I2C comm, but needed for not getting compile error

// TFT
#include <ESP_SSD1306.h>    // Modification of Adafruit_SSD1306 for ESP8266 compatibility
#include <Adafruit_GFX.h>   // Needs a little change in original Adafruit library (See README.txt file)
 

ESP_SSD1306 display(10);  //  

char LINELEN = 8;
char LINENUM = 4;

typedef struct  {
   char line[8][20] = {"line0_67 9012345678" ,"line1_6"      ,"line2_67" ,
                       "line3_6789012345678" ,"line4_6789012345678" ,"line5_6789012345678" ,
                       "line6_6789012345678" ,"line7_6789012345678" };
   char act=0, top=0, btm=top+LINENUM;
   char subl[12]="           ";
} list;

list mList00;

void strcpysubl(char * dest, char * source, char pos) {
   byte l;
   memset(dest,' ',sizeof(dest));
   strncpy( dest+1, source+pos, LINELEN );     
}

void  displayMenulist(list mList) {
  int8_t start;
  byte l;
  
  if(mList.act>LINENUM) start=mList.act;
  else start=0;
  
  for(byte i=0+start; i<5+start; i++) { 
    strcpysubl( mList.subl, mList.line[i], 0);     
      l=10 + i*13;
      display.setCursor( 0, l);  display.print(mList.subl);
      Serial.println(mList.subl);      
  } 
  display.display();
}  
                    

void setup(void)
{
 // Start Serial
 Serial.begin(115200);
 delay(2000); // wait for Serial()
  
 // Start Wire (SDA, SCL)
 Wire.begin(D2, D1);   

  // SSD1306 Init  
  display.begin(SSD1306_SWITCHCAPVCC);  // Switch OLED
  display.clearDisplay();  // Clear the buffer.
  
  display.setFont(); 
  displayMenulist(mList00);     
  
}

void loop() {
     
}

facchinm, 2018-12-03:
The sketch compiles cleanly with IDE master (so it also will with 1.8.7 since the builder didn't change) and esp8266 core 2.4.2.
Closing as solved in mainline.

closed 2018-12-03 by facchinm

dsyleixa, 2018-12-03:
thank you for your efforts, strangely I just now can't reproduce this issue either.

dsyleixa, 2018-12-03:
nonetheless, I am clueless - 
with the complete code it still don't work, unless I put the struct to the most upper position:
```

/*
 *  
 *********** This code is for the ESP8266 ***********
 * 
 ****** ESP_ssd1306_128x64_I2C demo display *********
 *
 *      using ESP_SSD1306 Adafruit library  
 * for a ssd1306 OLED Display  with the ESP8266 board
 *
 *********** This code is for the ESP8266 *********** 
 * 
*/

/*

  PINOUT:
  =======
  digital:       default      THIS          htmlStd       bRk            ESP_motorShield
   D0     16     WAKE         ---            ---          ---            ---
   D1      5     I2C SCL      I2C SCL       I2C SCL       I2C SCL        out D1 motorA
   D2      4     I2C SDA      I2C SDA       I2C SDA       I2C SDA        out D2 motorB
   D3      0     FLASH/LED    out D3/LED    out D3    built-in btn+LED   out D3 motorA 
   D4      2     TX1          in D4         In DHT        ---            out D4 motorB
   D5     14     SPI SCK      in D5         In B/S        out D5         SDA
   D6     12     SPI MISO     in D6         Out D6        ---            SCL
   D7     13     SPI MOSI     out D7        Out D7        out D7         in  D7 
   D8     15     MTD0 PWM     out D8        Out D8        ---            out D8
   D9      3     UART RX0     USB, Serial   USB, Serial   USB, Serial    USB, Serial
   D10     1     UART TX0     USB, Serial   USB, Serial   USB, Serial    USB, Serial
  (D12)   10     GPIO intern  GPIO intern   TFT reset     ---            TFT reset
*/

// i2c
#include <Wire.h>         // Incl I2C comm, but needed for not getting compile error
#define ESPSDA D2         // defaults
#define ESPSCL D1         // 

// TFT
//#include <Adafruit_SSD1306.h>
#include <ESP_SSD1306.h>    // Modification of Adafruit_SSD1306 for ESP8266 compatibility
#include <Adafruit_GFX.h>   // Needs a little change in original Adafruit library (See README.txt file)
#include <Fonts/FreeSans12pt7b.h>            // 
#include <Fonts/FreeSansBold12pt7b.h>        // 
#include <Fonts/FreeSans9pt7b.h>             //
#include <Fonts/FreeMono12pt7b.h>            //   
#include <Fonts/FreeMono9pt7b.h>    
// Pin definitions
#define OLED_RESET 10  // GPIO10=D12 Pin RESET signal (virtual)

//Adafruit_SSD1306 display(OLED_RESET);
//Adafruit_SSD1306 display(128, 64, &Wire, D0);
ESP_SSD1306 display(OLED_RESET); // FOR I2C

//----------------------------------------------------------------------------
// OLED dashboard
//----------------------------------------------------------------------------

void dashboard(byte mode) {     

  display.clearDisplay();
  display.setFont();

  if (mode == 0) {
    display.setFont();  // h=8.0 pt
    display.setCursor( 0,  0);  display.print(" 0 Hello World 1");
    display.setCursor( 0,  8);  display.print(" 8 Hello World 2");
[... 125 lines cut ...]
  display.setTextColor(WHITE);
  display.setCursor(0,0);
  display.println("Hello, world!");  
  display.display();

  delay(1000);
  display.clearDisplay(); 
  display.display();

  for(byte i=0; i<8; i++) { Serial.println(mList00.line[i]); }
  
  Serial.print("act="); Serial.println((byte)mList00.act);

  display.setFont(&FreeMono9pt7b); // h=13.10pt
  displayMenulist(mList00);
     
 // display.display();
  delay(2000);
  
}

void loop() {
   
}
```

dsyleixa, 2018-12-03:
thanks for your  efforts, too. 
Indeed I'm surprised how much the code could be cut down to reproduce the error considering that even the former code tested by facchinm was not able to show it.
Now I know what has happened for that: in order to cut it down before as much as possible, I deleted the unused dashboard function (belated, to make it more compact and better readable) which actually was crucial for producing  that compile error.
Having said that, I now see the issue clearly of course and how to fix that, thanks again.
I have to admit that migrating to newer IDE versions is not trivial because I cannot move my portable Windows version to my RaspberryPi easily with respect to my board installations, and reinstalling all and everything on either target platform (especially 3rd party boards like ESPs and Adafruit M0+M4+Huzzah  is really causing headaches, tbh.

per1234, 2018-12-03:
I'm not recommending you to migrate to the hourly or beta builds. The workaround of adding the function prototype is the best solution and that will work fine with your current Arduino IDE 1.8.5. There are some very nice improvements made in the Arduino IDE on every release and I think the upcoming 1.8.8 will be a good one but I do understand wanting to stick with the version that's working for you. We've already discussed your issue with transferring your installation between the Windows machine and RPi on the Arduino Forum.

However, as I said before, you should take the time to verify that bugs you find still occur in the hourly build before reporting them here. That doesn't mean you need to switch to using the hourly build full time. You can leave your existing installation in place. You only need to download and extract the hourly build, then run it long enough for a quick test. After that you can go back to whichever IDE version you are comfortable with.

Currently, the situation is a little confusing because there is an hourly build and a beta build. It is a bit onerous to need to download the latest version and test with both of those and so I feel that it's reasonable to test only with the hourly build. In this case the beta build is especially relevant since it uses a new sketch preprocessor. Hopefully soon the work in the beta build will get merged and we will be back to only having the hourly build to test with.

dsyleixa, 2018-12-03:
ok, I see that the hourly build is esasier to use for checking than I assumed, even if the 1.8.7 was supposed to produce the same error ;)
But I agree to your statements about testing of course, thx! 8)