arduino/Arduino #8259: library addition request: uStepper S-lite

uStepper, 2018-12-02:
Hi,

Please add the uStepper S-lite library, located at the following link:
https://github.com/uStepper/uStepper-S-lite
Thank you !

cmaglie, 2018-12-04:
Done

closed 2018-12-04 by cmaglie

per1234, 2018-12-13:
>nepustí nainštalovat rgbl dovload do arduino ide . prečo?

translation:
>does not let install rgbl dovload to arduino ide. Why?

@atilacsa please don't spam the issue tracker. The appropriate place to ask for help with your problem is the Arduino Forum. I'm sure we'll be able to help you there. 
Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules. Be sure to post enough information for us to help you. In this case, we would need the full error output.