arduino/Arduino #8260: HardwareSerial freezes on Arduino Pro Micro / Mac OS

iafan, 2018-12-02:
I have Arduino Pro Micro connected via USB to my Mac (Mac OS Sierra 10.12.6). Arduino IDE is 1.8.7, and the board is identified there as "Board: SparkFun Pro Micro / Processor: ATmega32U4 5V/16 Mhz". Board info gives: "BN: Unknown board / VID: 1B4F / PID: 9206".

If I send to it more than 384 bytes of data at once, only 384 bytes will be displayed, and the board will freeze, freezing Arduino Serial Monitor as well (or `screen`) until I unplug the board from USB. While frozen, I can't upload new sketch to it either.

Here's the simplest code that I use:

```cpp
void setup()
{
    Serial.begin(9600);
}

void loop()
{
    if (Serial.available())
    {
        Serial.write(Serial.read());
    }
}
```

So it reads the data at the fastest speed possible and just echoes it back. If I don't echo the code back, but just read it to empty the serial buffer, the problem still persists.

Once I upload this sketch to Arduino, I can either open a Serial Monitor, or use `screen /dev/cu.usbmodem14121 9600` from the console. As I type on the keyboard in `screen`, or send entire strings via Serial Monitors, the input will be echoed back. But If I copy a large block of text (more than 384 symbols) and paste that into the `screen` session or paste and send via Serial Monitor, only the first 384 symbols will be echoed back, and Arduino will freeze.

Rebooting the Mac doesn't help.

I noticed that if I force the serial speed to 1200bps (for example, by using `screen /dev/cu.usbmodem14121 1200`), then I will see no freezing. 2400bps and above results in the same freezing issue. I tested this on 3 different Arduino Pro Micro boards from different vendors, and all have the same issue. I used bare boards, with no external components attached to pins.

I understand that these boards connect USB directly to ATmega32U4 chip, so the problem seems to be either in the chip itself (less likely) or the USB drivers / HardwareSerial implementation in Arduino (more likely).

facchinm, 2018-12-03:
Hi @iafan ,
the `Serial` object exposed by boards based on 32u4 is USB based, so HardwareSerial is not involved. The code is [here](https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/CDC.cpp). Changing baud rate should not affect the CDC subsystem EXCEPT for 1200bps, which triggers a reboot to bootloader every time the port is opened with that baud. 
So the problem is always reproducible. I'd move the issue to https://github.com/arduino/ArduinoCore-avr if it was possible. @per1234 any idea why there's no button to move it :smile: ?

per1234, 2018-12-03:
I don't know why there's no transfer button. I don't see one either, but I assumed that was because I didn't have the necessary permission level in this repository. I'd assume you would. Maybe it's only available to repository Owners? That seems like an unnecessarily strict restriction.

There's another new feature that was released around the same time: [Delete issue](https://help.github.com/articles/deleting-an-issue/). I see that this feature [must be enabled at the organization level](https://help.github.com/articles/allowing-people-to-delete-issues-in-your-organization/). I wonder if the same applies to issue transfers but they forgot to document it?

I do have this button in my own repositories. The feature is in beta and I know sometimes GitHub does an opt-in for beta features but I don't remember whether I did that. I don't find any useful information on the [help article for this topic](https://help.github.com/articles/transferring-an-issue-to-another-repository/) or the [one about permission levels](https://help.github.com/articles/repository-permission-levels-for-an-organization/).

facchinm, 2018-12-03:
@mastrolinux would you mind taking a look if these :arrow_up:  options are enabled at organization level?

facchinm, 2018-12-04:
@iafan I'll appreciate if you could move it manually there and close this one, thanks! :slightly_smiling_face:

iafan, 2018-12-04:
Done!

closed 2018-12-04 by iafan

per1234, 2019-05-30:
>@mastrolinux would you mind taking a look if these ⬆️ options are enabled at organization level?

It turns out that GitHub's issue transfer feature was previously only available to those who had admin permissions in both repositories (though this restriction was not documented). Happily, GitHub changed it today so that you only need write permissions in the repositories:
https://github.blog/changelog/2019-05-29-improvements-to-issue-transfer-permissions/
So I now have a "Transfer issue" option in this repository's issues.

The restriction of only being able to transfer issues between repositories within an organization still applies (https://github.com/arduino/Arduino/issues/8245).