arduino/Arduino #8261: MacOS 10.14, IDE 1.9.0_Beta, Themes

a7hybnj2, 2018-12-02:
Just installed the 1.9 beta and it is awesome!

I did see in preferences the ability to change the theme. I looked into the theme folder and saw that there is a 'default' and a 'dark' theme in there. I am unable to see the dark theme as an available option in the IDE preferences.

Let me know if you want me to check or relay any addition configuration information.

facchinm, 2018-12-03:
Hi @a7hybnj2 ,
we are not yet shipping the Dark theme in the default installation but you have to install it yourself. Which `theme` folder are you referring to? To properly install the Dark theme in IDE 1.8.6+ you can simply copy the zip file into the `theme` folder inside sketchbook, then you can select if from Preferences.
[DarkArduino.zip](https://github.com/arduino/Arduino/files/2638713/DarkArduino.zip)

a7hybnj2, 2018-12-03:
Got it. Thanks. Was able to successfully install and switch between themes.

closed 2018-12-03 by a7hybnj2