arduino/Arduino #8262: Add Library to Library Manager

Fhandy, 2018-12-03:
Dear Arduino Admin,

Please add this library to Library Manager

https://git.antares.id/lorawan-loraid/arduino-loraid.git

cmaglie, 2018-12-04:
Done!

closed 2018-12-04 by cmaglie

Floessie, 2018-12-05:
@Fhandy I was quickly browsing your lib and found `LoRaIdClass::readData()` curious. Not only is it documented as `void read(void)` (which is why I was looking into it), but it also  takes a single pointer to be filled by the function. How can the caller know how big this buffer has to be without digging deeper? Better don't repeat the `strcpy` mess and add a `size_t` parameter and a public `constexpr size_t MAX_OUT_BUF_SIZE = 64` or `size_t getMaxOutputBufferSize() const`.

Best,
Flössie