arduino/Arduino #8271: Arduino verify in command line always fail caused by sketch folder and build.options.json

czgtest, 2018-12-08:
Repro on Arduino 1.8.7 and arduino-nightly-windows\arduino-nightly (latest Dec 6 )
OS : windows10

step:
1.download  Arduino 1.8.7 or arduino-nightly  
2.run verify command line

arduino_debug.exe --verify d:\test\test\1205\111\Blink_error\Blink_error\Blink.ino --board esp8266:esp8266:huzzah:CpuFrequency=80,VTable=flash,FlashSize=4M1M,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,FlashErase=none,UploadSpeed=115200 --pref build.path=d:\test\test\1205\111\Blink_error\Blink_error\build12

the first time , the verify is pass

[image]

run verify command in second time , it will fail
[image]

it always fail due to sketch folder and build.options.json not remoe.
[image]

[image]

facchinm, 2018-12-10:
@czgtest build folder is inside sketch folder, right? We are seeing some similar reports on https://github.com/arduino/arduino-builder ; as a first step, I'll go on and change the title of the issue. In the meantime, make sure to build outside the sketch folder to avoid triggering this bug.

czgtest, 2018-12-11:
[Blink_error.zip](https://github.com/arduino/Arduino/files/2666059/Blink_error.zip)

@facchinm   attach the folder to you

facchinm, 2018-12-19:
@czgtest I forgot that I already prepared a PR for this error :smile: 
Here it is https://github.com/arduino/arduino-builder/pull/299

I'm going ahead and merge it so we'll release a build of `arduino-builder` soonish with this improvement (and it will be available in IDE hourly)